lowRISC/lowrisc-chip #155: riscv environment variable

nak-tech, 2020-01-15:
I clone riscv-gnu-toolchain from https://github.com/riscv/riscv-gnu-toolchain, then install the toolschain, 
point RISCV :=/opt/riscv in Makefile,
the hierarchy in /opt/riscv is
/opt/riscv/
            |--bin
            |--include
            |--lib
            |--libexec
            |--riscv32-unknown-elf
            |--riscv64-unknown-linux-gnu
            |--share
            |--sysroot
then run make bitstream, I got following message:

make -C ../../../rocket-chip/vsim verilog
make[1]: Entering directory '/home/ghlab/work/lowrisc-chip/lowrisc-chip/rocket-chip/vsim'
/home/ghlab/work/lowrisc-chip/lowrisc-chip/rocket-chip/Makefrag:3: *** Please set environment variable RISCV. Please take a look at README. Stop.
make[1]: Leaving directory '/home/ghlab/work/lowrisc-chip/lowrisc-chip/rocket-chip/vsim'
Makefile:120: recipe for target 'verilog' failed
make: *** [verilog] Error 2

Wrong tool or wrong variable setting?

jrrk2, 2020-01-15:
try using:

export RISCV=/opt/riscv

in your shell. This will ensure that it applies to all commands you run, not just one Makefile. Likewise any settings in Makefiles will need export in front of them to apply to the environment of subsiduary Makefiles. Also be sue to mention what branch of lowrisc-chip you are working on in your issue.

jrrk2, 2020-01-16:
Did you install the list of pre-requisites as per the lowrisc documentation ?

sudo apt-get install autoconf automake autotools-dev curl \
libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison \
flex texinfo gperf libncurses5-dev libusb-1.0-0-dev libboost-dev \
swig git libtool libreadline-dev libelf-dev python-dev \
microcom chrpath gawk texinfo nfs-kernel-server xinetd pseudo \
libusb-1.0-0-dev hugo device-tree-compiler zlib1g-dev libssl-dev \
debootstrap debian-ports-archive-keyring qemu-user-static iverilog \
openjdk-8-jdk-headless iperf3 libglib2.0-dev libpixman-1-dev

nak-tech, 2020-01-16:
I did not see this information, is that should be found at  **documentation** link in
'See the documentation for build instructions.'
that is a void link.

nak-tech, 2020-01-16:
bit generated. Thanks.

closed 2020-01-16 by nak-tech