lowRISC/lowrisc-chip #162: Unexpected errors when building refresh-v0.6

PRieblinger-esatus, 2020-01-23:
## Environment

### PC
OS: Ubuntu Mate 18.04.3 LTS 64-bit
Kernel: Linux 5.0.0-37-generic x86_64
Memory: 8GB
CPU: Intel i5-2537M CPU, 1.4GHz x 4

### FPGA-board
Board Model: Nexys A7 
FPGA Model: Xilinx Artix-7 100T CSG324

### Development/Prototyping Environment
Xilinx Vivado v2019.2 (64-bit)

## What I am trying to accomplish
I want to run and demo Linux on an FPGA board. Alex Bradbury mentioned in [this talk](https://www.youtube.com/watch?v=emnN9p4vhzk) that a Nexys A7-100T was a good choice for this project, so I got one of these and installed the manufacturer's tools for developing and prototyping.
I had a somewhat hard time finding documentation on the subject, as the official lowrisc docs have not yet been migrated to the new site ( see #146 ) but eventually I found what I was looking for in the [lowrisc-site repository](https://github.com/lowRISC/lowrisc-site/).

## My Problem
According to the documentation on the old lowrisc-site, particularly [this](https://github.com/lowRISC/lowrisc-site/blob/master/content/docs/ethernet-v0.5/development.md), the lowrisc-chip can be built using the command `make bitstream`. The documentation refers to version ethernet-v0.5, not my version refresh-v0.6. I assumed it would still work, though.
When trying this on my system, the process eventually aborted with the following message:
```
Makefile:151: recipe for target 'lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.bit' failed
```
The reason for this seems to be the following errors:
```
*** Running vivado
    with args -log chip_top.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source chip_top.tcl

****** Vivado v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source chip_top.tcl -notrace
Command: synth_design -top chip_top -part xc7a100tcsg324-1 -flatten_hierarchy none
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7a100t'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a100t'
INFO: [Device 21-403] Loading part xc7a100tcsg324-1
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 13379 
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_addr' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:58]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_ba' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:59]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_ras_n' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:60]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_cas_n' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:61]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_we_n' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:62]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_ck_n' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:63]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_ck_p' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:64]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_cke' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:65]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_cs_n' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:66]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_dm' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:67]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'ddr_odt' when default_nettype is none [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:68]
INFO: [Synth 8-2350] module chip_top ignored due to previous errors [/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv:17]
Failed to read verilog '/home/prieblinger/git/lowrisc-chip/src/main/verilog/chip_top.sv'
```
I did try and research the issue but was not successful. Any help would be greatly appreciated, as I do not have a lot of experience with FPGAs.

jrrk2, 2020-01-23:
Dear Prieblinger,
  My apologies for the disruption to the website contents, this is outside of my control. Fortunately the content can be rendered locally to make it understandable. The documentation for the refresh-v0.6 release is in the main area of the website, not in a branch sub-directory:

git clone https://github.com/lowRISC/lowrisc-site.git
cd lowrisc-site
hugo server

The documentation you require is then accessible at http://localhost:1313/docs/
Referring to your main query, you should have regard for the comments on
 http://localhost:1313/docs/xilinx/
where it states that the tested version of Vivado is 2018.1 for this release.

If you don't have access to an earlier version of Vivado for some reason,
you can modernize the code by changing instances of 'output' to 'output wire'
to keep the tool happy.  E.G.:

 `elsif NEXYS4
   // DDR2 RAM
   inout wire [15:0]  ddr_dq,
   inout wire [1:0]   ddr_dqs_n,
   inout wire [1:0]   ddr_dqs_p,
   output [12:0] ddr_addr,
   output [2:0]  ddr_ba,

should be changed to:

 `elsif NEXYS4
   // DDR2 RAM
   inout wire [15:0]  ddr_dq,
   inout wire [1:0]   ddr_dqs_n,
   inout wire [1:0]   ddr_dqs_p,
   output _**wire**_ [12:0] ddr_addr,
   output _**wire**_ [2:0]  ddr_ba,

and so on.

The reason for the tool version dependence is that Vivado has an automatic algorithm for selecting the order of compiling modules. but directives such as default_nettype apply globally to all modules after that directive.

PS the ariane-v0.7 release will be out soon. If you are interested in X-Windows and/or GenesysII board support and the option to support the Ariane SystemVerilog CPU you can check out the docs on the ariane-v0.7 branch.

PRieblinger-esatus, 2020-01-23:
Thank you very much for your quick and helpful response!
The changes from "output" to "output wire" in each line that caused an error resolved the problem. The command `make bitstream` just completed successfully, as well as the following commands `make cfgmem` and `make program-cfgmem`.
Using Hugo to open the documentation did not work using the version of Hugo in the Ubuntu 18.04 LTS repo, but it did using the snap version. I also mentioned this [in issue #146](https://github.com/lowRISC/lowrisc-chip/issues/146#issuecomment-577653058).

closed 2020-01-23 by PRieblinger-esatus