lowRISC/lowrisc-chip #169: how to add new peripheral in Ariane-v0.7

zlt323, 2020-03-23:
Thank you! Is there a demo showing how to add new peripheral? Such as Gpio.

jrrk, 2020-03-23:
No, but you can copy or customise the functionality connected to the switches.

zlt323, 2020-03-23:
Ok, I will have a try.

zlt323, 2020-03-23:
When I am starting Debian, there are some error, and I can't connect to the Internet

[FAILED] Failed to start Raise network interfaces.
See 'systemctl status networking.service' for details.
[  OK  ] Reached target Network.
[  OK  ] Reached target Network is Online.
         Starting ATA over Ethernet discovery...
[  OK  ] Started ATA over Ethernet discovery.
[  OK  ] Started Create Volatile Files and Directories.
         Starting RPC bind portmap service...
         Starting Network Time Synchronization...
         Starting Update UTMP about System Boot/Shutdown...
[  OK  ] Started RPC bind portmap service.
[  OK  ] Reached target Remote File Systems (Pre).
[  OK  ] Reached target Remote File Systems.
[  OK  ] Reached target RPC Port Mapper.
[  OK  ] Started Update UTMP about System Boot/Shutdown.
[  OK  ] Started Network Time Synchronization.
[  OK  ] Reached target System Initialization.
[  OK  ] Started Daily Cleanup of Temporary Directories.
[  OK  ] Reached target System Time Set.
[  OK  ] Reached target System Time Synchronized.
[FAILED] Failed to start Daily apt download activities.
See 'systemctl status apt-daily.timer' for details.
[FAILED] Failed to start Daily apt upgrade and clean activities.
See 'systemctl status apt-daily-upgrade.timer' for details.
[  OK  ] Started Periodic ext4 Onli…ata Check for All Filesystems.
[  OK  ] Started Daily exim4-base housekeeping.
[  OK  ] Started Daily rotation of log files.
[FAILED] Failed to start Daily man-db regeneration.

zlt323, 2020-03-23:
I found isc-dhcp-clien was not intsall correctly. Maybe that is reason for this error

jrrk, 2020-03-23:
Is-dhcp-client is not needed because it’s functionality is built into systemd

jrrk, 2020-03-23:
I recommend the buildroot option, Debian is a bit unstable just now

zlt323, 2020-03-23:
Ok, before debian, I tried the buildroot. It's very simple and fast. However, it's not very convenient to install software. Besides compiling the source code, is there any easier way to install the software, such as Python.

jrrk, 2020-03-23:
Python, Perl, Java etc are all interpreters which run slowly on the FPGA and are not (yet) optimised for RISCV. The intended buildroot usage is to add your software to the buildroot tree as needed. This release of buildroot does not support compilation on the target. You are free to install Python but it will run rather slowly and use more memory than a C/C++ program.

zlt323, 2020-03-23:
Thank you for your help, there are so much knowledge I need learn, and I will spend more time on it.

zlt323, 2020-03-23:
How should I use X-windows? I change the ~/.xinitrc and get the result:

 Error: Can't open display: 
xterm: Xt error: Can't open display: 
xterm: DISPLAY is not set

jrrk, 2020-03-23:
Take a look at the X-windows page in the LowRISC documentation. You need a Bluetooth mouse and PMOD-BT adapter. The command startx run on the VGA console starts the server.

zlt323, 2020-03-27:
How can I compile C++ program in Buildroot system, there is not gcc or g++. 
I have try to cross-compile my program, but it will report an error if I call function such as opencv;

jrrk, 2020-03-27:
Check out these instructions:

https://buildroot.org/downloads/manual/manual.html#adding-packages

If your code uses some obscure functionality, you can enable it (if supported) in the buildroot configuration file.

zlt323, 2020-03-27:
I copy the file 'mainf/host' to my  home file. Then successfully compile the c++ program using gcc(add command '-I ~/host/include') in '/host/bin'.
I have ever tried to add software using buildroot, but it seems that I must recompile all the packages which is too slow.

zlt323, 2020-03-27:
I try to using gdb to debug my new gpio, but I find it is useless to change my main.c program and cpu can't read my new variable.

jrrk, 2020-03-27:
You can’t use gdb to debug Linux applications at the moment on this platform, only the kernel itself or bare metal.

zlt323, 2020-03-27:
Sorry, I mean I'm doing bare metal testing and I've modified ‘/lowrisc-chip-ariane-v0.7/fpga/src/etherboot/src/main.c’. How can I compile the main.c to debug my hardware.

jrrk, 2020-03-27:
You just run make in the etherboot directory, or you can try ‘make firmware’ in the top level.

zlt323, 2020-03-27:
Thank you very much, it helps me a lot.

FloyedShen, 2020-03-31:
Hello, I want to add some linux device drivers, but I find /usr/src is empty. How should I get the required linux-headers?

jrrk, 2020-03-31:
In this release, device drivers for Linux are managed by buildroot as a patch to a stock Linux download. If you want to develop your own drivers, you could either copy the tree out of buildroot after automatic patching or download your own copy of the kernel and patch it manually. Either way /use/src is not used, but you could use it you want the same Linux kernel to run on your desktop.

FloyedShen, 2020-04-01:
Thank you. I tried to compile the GPIO driver on the host using the toolchain and kernel provided by buildroot. But when I tried to mount the driver on the kernel, I got some errors.
```
root@lowrisc:/home/test# insmod gpio.ko
[  430.244004] demo: loading out-of-tree module taints kernel.
../machine/mtrap.c:21: machine mode: unhandlable trap 4 @ 0x0000000080004c28
Power off
```
This is very confusing. Do you have any idea to solve this problem? Or I can only compile the driver into the kernel.

FloyedShen, 2020-04-01:
Thank you. I tried to compile the GPIO driver on the host using the toolchain and kernel provided by buildroot. But when I tried to mount the driver on the kernel, I got some errors.

```
root@lowrisc:/home/test# insmod gpio.ko
[  430.244004] gpio: loading out-of-tree module taints kernel.
../machine/mtrap.c:21: machine mode: unhandlable trap 4 @ 0x0000000080004c28
Power off
```

This is very confusing. Do you have any idea to solve this problem?

jrrk, 2020-04-01:
If you compile the kernel in monolithic mode (not using modules) then it is a lot easier to locate the program counter for any errors from the kernel disassembly view.

jrrk, 2020-04-01:
You can identify trap 4 from the exception codes in the privileged instructions manual.

zlt323, 2020-04-21:
> Thank you. I tried to compile the GPIO driver on the host using the toolchain and kernel provided by buildroot. But when I tried to mount the driver on the kernel, I got some errors.
>
> ``` [...]

Have you solved this problem successfully?

jrrk, 2020-04-21:
I have not tested using modules at all. I always build the kernel monolithic. This saves issues with tainted messages.

jrrk, 2020-04-21:
.. also if you build monolithic, you can relate the error program counter more easily to the kernel disassembly.

zlt323, 2020-04-21:
Thank, I have solved this problem successfully just now.

zlt323, 2020-04-26:
I am trying to using dma to move data from DDR to my new peripheral, and there are some problems. 
I think I need to configuring the DDR controller to have a second AXI slave port. Should I connect it to DMA's AXI port  or AXI BUS?
If I connect it to AXI_BUS,  the DMA works as a master, but I find  there are only one master port in 'demux' and it was be connected to Rocket.

jrrk, 2020-04-26:
You can try to add a mux module before the demux, but this is outside the scope of what mailing list support can provide.

zlt323, 2020-04-26:
Thank you, I will have a try.

LeooLevinn, 2020-06-08:
> Thank you, I will have a try.

you must be doing the RISCV挑战杯 dude. I also have the problem "failed to raise network interface", how did you solve it?