lowRISC/lowrisc-chip #170: Login incorrect in Ariane-v0.7

Suedongchu, 2020-04-02:
[  OK  ] Stopped Serial Getty on hvc0.
[  OK  ] Started Serial Getty on hvc0.

Welcome to Buildroot
buildroot login: k
Password:
Login incorrect
buildroot login:

I don't know the password
Is there a way like v0.6's firstboot.riscv to set passwd?

FloyedShen, 2020-04-02:
I think you can clear the password by modifying `/etc/shadow`

jrrk, 2020-04-02:
The v0.7 release has moved to buildroot which manages the initial root password with a configuration file. If you haven’t changed that file you should be able to login with username root and password LowRISC (if I recall correctly).

jrrk, 2020-04-02:
You can then use passwd lowrisc to set a password for the default account, which by default does not require a password (but ssh is disabled without a password)

Suedongchu, 2020-04-02:
Thanks for yours‘ information
I want to ask another one
When I use the repo lowrisc-quickstart v0.6 to make my microsd into 4 partitions, I encountered problem while booting Fedora：

[ TIME ] Timed out waiting for device /dev/mmcblk0p4.
[DEPEND] Dependency failed for File…ystem Check on /dev/mmcblk0p4.
[DEPEND] Dependency failed for /home.
[DEPEND] Dependency failed for Local File Systems.
[ TIME ] Timed out waiting for device /dev/mmcblk0p1.
[DEPEND] Dependency failed for /mnt/dos.
[DEPEND] Dependency failed for File…ystem Check on /dev/mmcblk0p1.
[ TIME ] Timed out waiting for device /dev/mmcblk0p3.
[DEPEND] Dependency failed for /dev/mmcblk0p3.
[DEPEND] Dependency failed for Swap.

it seems the linux cannot find the sdcard.
I had ever changed the devices' name in /etc/fstab into sdc/UUID... but all failed.
I don't even know what the problem is.

jrrk, 2020-04-02:
The v0.7 release implements the SD block layer in hardware using a completely different driver which uses rda1,2,3,4 as its names. You can find updated quickstart scripts as a sub-module of the lowrisc-chip repo. You will find Fedora is rather heavyweight on this platform, but it does boot, sort of.

Suedongchu, 2020-04-04:
Thanks @jrrk 
emm ..
I encountered problem again in v0.6's debian.
For some reasons, I  want to use the ftp service in debian.
so I just login as root and then :
root@lowrisc:/# apt update && apt install ftp
There came the problem:

Get:1 http://ftp.ports.debian.org/debian-ports sid InRelease [55.3 kB]
Get:2 http://ftp.ports.debian.org/debian-ports unreleased InRelease [56.6 kB]
Err:1 http://ftp.ports.debian.org/debian-ports sid InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 84C573CD4E1AFD6C
Err:2 http://ftp.ports.debian.org/debian-ports unreleased InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 84C573CD4E1AFD6C
Fetched 112 kB in 25s (4,438 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://ftp.ports.debian.org/debian-ports sid InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 84C573CD4E1AFD6C
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://ftp.ports.debian.org/debian-ports unreleased InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 84C573CD4E1AFD6C
W: Failed to fetch http://ftp.ports.debian.org/debian-ports/dists/sid/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 84C573CD4E1AFD6C
W: Failed to fetch http://ftp.ports.debian.org/debian-ports/dists/unreleased/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 84C573CD4E1AFD6C
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  ftp
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 57.3 kB of archives.
After this operation, 124 kB of additional disk space will be used.
Err:1 http://ftp.ports.debian.org/debian-ports sid/main riscv64 ftp riscv64 0.17-34
  404  Not Found [IP: 149.20.4.15 80]
E: Failed to fetch http://ftp.ports.debian.org/debian-ports/pool-riscv64/main/n/netkit-ftp/ftp_0.17-34_riscv64.deb  404  Not Found [IP: 149.20.4.15 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

I google a lot about gpg and got some solution like this:
gpg --keyserver keyring.debian.org --recv-keys 84C573CD4E1AFD6C
gpg --armor --export 84C573CD4E1AFD6C | apt-key add -

gpg: directory '/root/.gnupg' created
gpg: keybox '/root/.gnupg/pubring.kbx' created
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0

Then I kept looking for the reason and solution until now for 2 days. 
Can you help me. Thank you for your patience.

jrrk, 2020-04-04:
The Debian repository has changed quite a bit since the v0.6 release. I would base your download on the .7 release and add any necessary packages to the install script for now. Of course most people just use wget or rsync instead of ftp nowadays.

Debian installs are not always stable which is why we recommend buildroot for the latest release.

Suedongchu, 2020-04-04:
Thank you@jrrk
I got your point.
I have used the buildroot of v0.7. But I can't find the gcc or g++ .How can i install such services?

jrrk, 2020-04-04:
For some reason buildroot discontinued compiler support several releases earlier. I suppose you could port the compiler build scripts from an earlier buildroot release, or see if the compilers from Fedora RISCV stage4 could work. You could unpack a foreign release in a chroot, or adapt it as your main release (but the trick I use to support X-windows on Debian and buildroot doesn’t work on Fedora-it is too new)

It is not advisable to install anything too heavyweight because the SD-Card has limited life and reliability.

Also, you only have 128M of RAM on the Nexys4ddr so more modern distributions will struggle.

Another alternative is to just add whatever you want to compile to buildroot as a local package.

Regards,
Jonathan