lowRISC/lowrisc-chip #171: make fail in version0.7

jjyy-Huang, 2020-04-03:
Hello, I am trying to create in version 0.7, but it failed. It stopped in building host-qemu 3.1.1.1.

> root@ubuntu:/opt/lowrisc-chip-0.7# make
make -C buildroot-2019.11.1-lowrisc/mainfs
make[1]: Entering directory '/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs'
host-qemu 3.1.1.1 Building
unset TARGET_DIR; PATH="/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin:/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/host/sbin:/opt/lowrisc-chip-0.7/tools/bin:/opt/Xilinx/SDK/2018.1/bin:/opt/Xilinx/SDK/2018.1/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.1/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.1/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.1/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.1/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.1/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/opt/Xilinx/Vivado/2018.1/bin:/opt/Xilinx/SDK/2018.1/bin:/opt/lowrisc-chip-0.6/riscv/bin" PKG_CONFIG="/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/host/lib/pkgconfig:/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/host/share/pkgconfig" /usr/bin/make -j5 -C /opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1
	CHK version_gen.h
  LINK    qemu-nbd
/usr/bin/ld: cannot find -lxml2
collect2: error: ld returned 1 exit status
/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/rules.mak:124: recipe for target 'qemu-nbd' failed
make[3]: *** [qemu-nbd] Error 1
package/pkg-generic.mk:238: recipe for target '/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_built' failed
make[2]: *** [/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_built] Error 2
Makefile:23: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs'
Makefile:175: recipe for target '/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar' failed
make: *** [/opt/lowrisc-chip-0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar] Error 2

How can i make it. And where i can find the guidance documents.

Thanks !

jrrk, 2020-04-03:
The instructions are here:

https://www.cl.cam.ac.uk/~jrrk2/docs/docs/

It looks like the build is failing because libxml2-dev is not installed.

jjyy-Huang, 2020-04-03:
Thank you for your reply !
I have found my mistake, maybe i miss apt something. Now it run again.
Thanks again !

jjyy-Huang, 2020-04-03:
And now i have a new issue. When i `make nexys4_ddr_rocket`, it seems something wrong in Makefile.

> root@ubuntu:/opt/lowrisc-chip-0.7# make nexys4_ddr_rocket
make fpga/work-fpga/nexys4_ddr_rocket/rocket_xilinx.mcs BOARD="nexys4_ddr" CPU="rocket" BITSIZE=0x400000 XILINX_PART="xc7a100tcsg324-1" XILINX_BOARD="digilentinc.com:nexys4_ddr:part0:1.1" VENDOR="sifive" MEMSIZE="0x8000000" BBL=/opt/lowrisc-chip-0.7/lowrisc-quickstart/rescue.bin |& tee nexys4_ddr_rocket.log
/bin/sh: 1: Syntax error: "&" unexpected
Makefile:62: recipe for target 'nexys4_ddr_rocket' failed
make: *** [nexys4_ddr_rocket] Error 2

What should i do ?
Looking forward to your reply.

jrrk, 2020-04-03:
/bin/sh should be a link to /bin/bash rather than /bin/dash

jjyy-Huang, 2020-04-04:
Hello, doctor @jrrk . Thank you for your reply.
Now I can program my neyxs4ddr and use debian by SDcard. But when i try to use gpio to light a LED (it's necessary for a freshman),  but /sys/class/gpio is empty. What should I do next.
I am grateful for your patient guidance.

jrrk, 2020-04-04:
That entry appears when a suitable GPIO device driver is loaded. Perhaps your professor was expecting you to write that driver (which seems a bit mean for a freshman exercise) or adapt it from an earlier version of LowRISC. Either way you’ve got some learning to do and it would not be in your best interests for me to just give you the solution. If I were you I might start with a simulator such as spike or QEMU before trying it on the FPGA board. I might be able to comment further if I knew what University and course you were on and what the syllabus is.

jjyy-Huang, 2020-04-04:
Thank you for your reply, doctor @jrrk .
I came into contact with this board and this open source project by chance. I have learned the basics of stm32 before, so now I want to further study the development and study of Linux system. This is just my interest. I will take your advice that write my own driver.
Thanks again.

closed 2020-05-02 by jjyy-Huang