lowRISC/lowrisc-chip #172: Ariane-v0.7 Failed to generate bitstream for nexys4ddr

LiCheng086, 2020-04-15:
HI. I was following all the instruction on the websitehttps://www.cl.cam.ac.uk/~jrrk2/docs/generate-the-bitstream/, I use the instruction"make nexys4_ddr_rocket_program", but it failed with these errors
[image]
[image]
I am working with ubuntu16.04LTS and vivado2018.1. I have no idea what went wrong, could some one help me? I am really appreciate your help.

LiCheng086, 2020-04-15:
error messages are shown here:
LINK    qemu-io
/usr/bin/ld: cannot find -lxml2
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x249): warning: Using 'getpwuid' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0xcf): warning: Using 'getpwnam_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x106): warning: Using 'getpwuid_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
io/dns-resolver.o: In function `qio_dns_resolver_lookup_sync_inet':
/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/io/dns-resolver.c:89: warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
collect2: error: ld returned 1 exit status
/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/rules.mak:124: recipe for target 'qemu-nbd' failed
make[3]: *** [qemu-nbd] Error 1
make[3]: *** Waiting for unfinished jobs....
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x249): warning: Using 'getpwuid' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0xcf): warning: Using 'getpwnam_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x106): warning: Using 'getpwuid_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/bin/ld: cannot find -lxml2
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x249): warning: Using 'getpwuid' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0xcf): warning: Using 'getpwnam_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x106): warning: Using 'getpwuid_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
io/dns-resolver.o: In function `qio_dns_resolver_lookup_sync_inet':
/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/io/dns-resolver.c:89: warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
collect2: error: ld returned 1 exit status
/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/rules.mak:124: recipe for target 'qemu-io' failed
make[3]: *** [qemu-io] Error 1
make[3]: *** wait: No child processes.  Stop.
package/pkg-generic.mk:238: recipe for target '/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_built' failed
make[2]: *** [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_built] Error 2
Makefile:23: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs'
Makefile:175: recipe for target '/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar' failed
make: *** [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar] Error 2

jrrk, 2020-04-15:
libxml2-dev is not installed

LiCheng086, 2020-04-16:
Thanks, however, another error appeared after I installed libxml2-dev is not installed:
[image]
What should I do next?

jrrk, 2020-04-16:
Try changing your system shell from dash to bash:

ls -l /bin/sh

LiCheng086, 2020-04-16:
I've tried, and here comes new errors and warnings:
`[info] downloading https://repo1.maven.org/maven2/org/jruby/jruby-complete/1.7.21/jruby-complete-1.7.21.jar ...
sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1[warn] 	[FAILED     ] org.jruby#jruby-complete;1.7.21!jruby-complete.jar(bundle): timeout (1801597ms)
[warn] 	Detected merged artifact: [FAILED     ] org.jruby#jruby-complete;1.7.21!jruby-complete.jar(bundle):  (0ms).
[warn] ==== typesafe-ivy-releases: tried
[warn] ==== sbt-plugin-releases: tried
[warn]   https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/org.jruby/jruby-complete/1.7.21/bundles/jruby-complete.jar
[warn] ==== local: tried
[warn]   /home/richard/.ivy2/local/org.jruby/jruby-complete/1.7.21/bundles/jruby-complete.jar
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/org/jruby/jruby-complete/1.7.21/jruby-complete-1.7.21.jar
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/richard/.sbt/preloaded/org.jruby/jruby-complete/1.7.21/bundles/jruby-complete.jar
[warn] ==== local-preloaded: tried
[warn]   file:////home/richard/.sbt/preloaded/org/jruby/jruby-complete/1.7.21/jruby-complete-1.7.21.jar
[warn] ==== scalasbt: tried
[warn]   http://scalasbt.artifactoryonline.com/scalasbt/sbt-plugin-releases/org.jruby/jruby-complete/1.7.21/bundles/jruby-complete.jar
[warn] ==== sbt-plugin-releases: tried
[warn]   https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/org.jruby/jruby-complete/1.7.21/bundles/jruby-complete.jar
[warn] ==== jgit-repo: tried
[warn]   http://download.eclipse.org/jgit/maven/org/jruby/jruby-complete/1.7.21/jruby-complete-1.7.21.jar
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::              FAILED DOWNLOADS            ::
[warn] 	:: ^ see resolution messages for details  ^ ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: org.jruby#jruby-complete;1.7.21!jruby-complete.jar(bundle)
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[error] sbt.librarymanagement.ResolveException: download failed: org.jruby#jruby-complete;1.7.21!jruby-complete.jar(bundle)
[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:331)
[error] 	at sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1(IvyActions.scala:205)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.$anonfun$withModule$1(Ivy.scala:243)
[error] 	at sbt.internal.librarymanagement.IvySbt.$anonfun$withIvy$1(Ivy.scala:204)
[error] 	at sbt.internal.librarymanagement.IvySbt.sbt$internal$librarymanagement$IvySbt$$action$1(Ivy.scala:70)
[error] 	at sbt.internal.librarymanagement.IvySbt$$anon$3.call(Ivy.scala:77)
[error] 	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:95)
[error] 	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:80)
[error] 	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:99)
[error] 	at xsbt.boot.Using$.withResource(Using.scala:10)
[error] 	at xsbt.boot.Using$.apply(Using.scala:9)
[error] 	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:60)
[error] 	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:50)
[error] 	at xsbt.boot.Locks$.apply0(Locks.scala:31)
[error] 	at xsbt.boot.Locks$.apply(Locks.scala:28)
[error] 	at sbt.internal.librarymanagement.IvySbt.withDefaultLogger(Ivy.scala:77)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:199)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:196)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.withModule(Ivy.scala:242)
[error] 	at sbt.internal.librarymanagement.IvyActions$.updateEither(IvyActions.scala:190)
[error] 	at sbt.librarymanagement.ivy.IvyDependencyResolution.update(IvyDependencyResolution.scala:20)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:56)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:46)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:99)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:68)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$19(LibraryManagement.scala:112)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:224)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:112)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:95)
[error] 	at sbt.util.Tracked$.$anonfun$inputChanged$1(Tracked.scala:149)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:126)
[error] 	at sbt.Classpaths$.$anonfun$updateTask$5(Defaults.scala:2383)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:39)
[error] 	at sbt.std.Transform$$anon$4.work(System.scala:66)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:262)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error] 	at sbt.Execute.work(Execute.scala:271)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:262)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:174)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:36)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[... cut ...]
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (update) sbt.librarymanagement.ResolveException: download failed: org.jruby#jruby-complete;1.7.21!jruby-complete.jar(bundle)
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? 
`
Dose it happened bacause I didn't install something else? Looking forward to your reply. Thank you!

jrrk, 2020-04-16:
Not seen this before, it could be because an incompatible version of Java is installed compared to our recommended package. You should still be able to run the ariane build, however.

LiCheng086, 2020-04-17:
Thank you! I chose the "ignore", and change the vivado version in fpga/Makefile. It started to synthesis but failed with errors:
[image]
followed by all the sourse files
[image]
following these errors and warnings:

> Top: instr_trace_item
WARNING: [filemgmt 56-315] Source scanning failed during design analysis. To get more details run synthesis or simulation and check the log.
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/src/OpenIP/axi/channel.sv:53]
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/src/OpenIP/axi/channel.sv:55]
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/src/OpenIP/util/fifo.sv:53]
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/src/OpenIP/util/fifo.sv:54]
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/src/OpenIP/axi/lite_buf.sv:42]
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/src/OpenIP/axi/lite_channel.sv:48]
WARNING: [Synth 8-1921] elaboration system task fatal violates IEEE 1800 syntax [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/src/OpenIP/axi/lite_join.sv:36]
WARNING: [Synth 8-2841] use of undefined macro PHY_ADDR_WIDTH [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_cache_manager.sv:55]
WARNING: [Synth 8-2841] use of undefined macro MSG_TYPE [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:109]
ERROR: [Synth 8-2496] packed dimension must specify a range [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:109]
ERROR: [Synth 8-2715] syntax error near type [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:109]
WARNING: [Synth 8-2841] use of undefined macro MSG_LENGTH [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:110]
ERROR: [Synth 8-2496] packed dimension must specify a range [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:110]
WARNING: [Synth 8-2841] use of undefined macro MSG_MSHRID [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:111]
ERROR: [Synth 8-2496] packed dimension must specify a range [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:111]
WARNING: [Synth 8-2841] use of undefined macro MSG_DATA_SIZE_ [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:113]
ERROR: [Synth 8-2496] packed dimension must specify a range [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:113]
WARNING: [Synth 8-2841] use of undefined macro MSG_SRC_CHIPID_ [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:114]
ERROR: [Synth 8-2496] packed dimension must specify a range [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:114]
WARNING: [Synth 8-2841] use of undefined macro MSG_SRC_X_ [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:115]
ERROR: [Synth 8-2496] packed dimension must specify a range [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:115]
WARNING: [Synth 8-2841] use of undefined macro MSG_SRC_Y_ [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:116]
ERROR: [Synth 8-2496] packed dimension must specify a range [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:116]
ERROR: [Synth 8-2715] syntax error near == [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:122]
WARNING: [Synth 8-2841] use of undefined macro MSG_TYPE_STORE_REQ [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:122]
WARNING: [Synth 8-2841] use of undefined macro MSG_TYPE_STORE_MEM [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:123]
ERROR: [Synth 8-2715] syntax error near == [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:124]
WARNING: [Synth 8-2841] use of undefined macro MSG_TYPE_NC_STORE_REQ [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:124]
ERROR: [Synth 8-2715] syntax error near == [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:126]
WARNING: [Synth 8-2841] use of undefined macro MSG_TYPE_LOAD_REQ [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:126]
WARNING: [Synth 8-2841] use of undefined macro MSG_TYPE_LOAD_MEM [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:127]
[... cut ...]
ERROR: [Synth 8-2715] syntax error near <= [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:220]
ERROR: [Synth 8-2715] syntax error near ] [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:220]
ERROR: [Synth 8-2715] syntax error near ; [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:220]
ERROR: [Synth 8-2715] syntax error near ] [/home/richard/Work/github/lowrisc-chip-ariane-v0.7/fpga/src/noc_sd_bridge/rtl/piton_sd_core_ctrl.sv:221]

[image]

jrrk, 2020-04-17:
I do not recommend changing Vivado version, I said Java version might be wrong. Please see if the ariane build target works, this is independent of Java/Scala.

LiCheng086, 2020-04-20:
@jrrk Thank you! It worked and generated bitstream. However, when I follow the "Flash the SD card" and execute the instruction`make sdcard-install USB=sdc`, it failed with errors:
[image]

jrrk, 2020-04-20:
It looks like building or downloading the rootfs failed for some reason. Try deleting and recreating or redownloading.

LiCheng086, 2020-04-21:
@jrrk 
Thank you! After rebuilt the rootfs, there is no warnings anymore. I plug the sd card into the board, the board started to boot. However, it faild to read elf, here are messages through microcom:

> Hello World!
Setup MAC addr
SD boot
    sd_f:  0x0  dma_f: 0x0 status: 
    resp_vec: 0x0  init_state: 0xa  counter: 0x562
    init_fsm: 0x1  tran_state: 0x3f  tran_fsm: 0x0
    sd_f:  0x0  dma_f: 0x0 status: 
    resp_vec: 0x0  init_state: 0xa  counter: 0x72602
    init_fsm: 0x1  tran_state: 0x3f  tran_fsm: 0x0
    sd_f:  0x0  dma_f: 0x0 status: 
    resp_vec: 0x0  init_state: 0x14  counter: 0x0
    init_fsm: 0x0  tran_state: 0x3f  tran_fsm: 0x0
    sd_f:  0x0  dma_f: 0x0 status: 
    resp_vec: 0x0  init_state: 0x47  counter: 0x3ab61
    init_fsm: 0x1  tran_state: 0x3f  tran_fsm: 0x0
    sd_f:  0x0  dma_f: 0x0 status: 
    resp_vec: 0x0  init_state: 0x31  counter: 0x0
    init_fsm: 0x0  tran_state: 0x3f  tran_fsm: 0x0
    sd_f:  0x0  dma_f: 0x0 status:  REQ_RDY INIT_DONE HCXC
    resp_vec: 0x0  init_state: 0xf0  counter: 0x0
    init_fsm: 0x0  tran_state: 0x0  tran_fsm: 0x0
    sd_f:  0x0  dma_f: 0x0 status:  REQ_RDY INIT_DONE HCXC
    resp_vec: 0x0  init_state: 0xf0  counter: 0x0
    init_fsm: 0x0  tran_state: 0x0  tran_fsm: 0x0
Load boot.bin into memory
load elf to DDR memory
len required = 40, actual = 9
elf read failed with code 2

What does code2 means and what should I do next?

jrrk, 2020-04-21:
I wonder if you should try a different sdcard (as recommended in the docs) or maybe the QSPI or Ethernet boot option.

LiCheng086, 2020-04-21:
Thank you for your patience! It worked when I am using QSPI to boot. Thank you very much! @jrrk

closed 2020-04-21 by LiCheng086