lowRISC/lowrisc-chip #174: Can‘t print Boot log by quick start

chunxue-ding, 2020-06-12:
Hi，I use quick start to launch Nexys4ddr board ，When I finish the instructions ——make USB=sdc mkfs fatdisk extdisk，and Unplug the SD card, insert the SD card slot, turn on the board power switch, the busy light is flickering slowly,JP1 is USB/SD JP2 is sd，but  it still hasn't started ，nothing  happened

jrrk, 2020-06-12:
This release doesn’t support configure from SD.

chunxue-ding, 2020-06-12:
Thank you  for answer my question sir,I try to use Remote console installation to complete the installation ,but when i  finish the instructions —— make memstick USB=sdc ,I found only one .bit file on my SD card,and this file only 9 KB, Open text content  is   not found

jrrk, 2020-06-14:
Perhaps your download failed for some reason. Try again after a make clean. And be sure to include a log of everything that happened if you need help, the important message may not be what you think.

chunxue-ding, 2020-06-15:
Thank you for you advice sir , There is indeed an error in the downloaded file.After downloading the correct file，Start printing  Boot log，But he made a mistake in the open boot ，after print Load boot.bin into memory ,it print **Failed to open boot!** ,hope for you advice ,thank you very much .

chunxue-ding, 2020-06-15:
[image]

zGitHubj, 2020-06-16:
Maybe, you can use the command make USB=sdc sdcard.img. But you must use the commad sh cardmem.sh before that.
You can verify the Makefile and cardmem.sh to customize the SD card.

chunxue-ding, 2020-06-17:
I have tried the above methods, but it is still such a mistake

zGitHubj, 2020-06-17:
Please contact me via zjerry940325810@gmail.com and share your some specific mistakes.

closed 2020-06-22 by chunxue-ding