lowRISC/lowrisc-chip #175: Makefile errors(failed to make nexys4_ddr_rocket with rc2)

LeooLevinn, 2020-06-27:
I noticed that maybe there are some errors in 0.7rc2 makefile commands.
I step-by-step followed the intro at https://www.cl.cam.ac.uk/~jrrk2/docs/generate-the-bitstream/
At the step of generate the bitstream, some error occurs.
I run the command: make nexys4_ddr_rocket
Then, it was running until an error shows:

> make fpga/work-fpga/nexys4_ddr_rocket/rocket_xilinx.mcs BOARD="nexys4_ddr" CPU="rocket" BITSIZE=0x400000 XILINX_PART="xc7a100tcsg324-1" XILINX_BOARD="digilentinc.com:nexys4_ddr:part0:1.1" VENDOR="sifive" MEMSIZE="0x8000000" BBL=/home/leo/makess/lowrisc-chip-ariane-v0.7/lowrisc-quickstart/rescue.bin |& tee nexys4_ddr_rocket.log
/bin/sh: 1: Syntax error: "&" unexpected
Makefile:61: recipe for target 'nexys4_ddr_rocket' failed
make: *** [nexys4_ddr_rocket] Error 2

I thought there may be some mistakes in the Makefile, so I deleted the "&" after "|". Then, I run the make nexys4_ddr_rocket again. This time, the |& thing was OK, but another error occurs:

> sed -e "s/VENDOR/sifive/" -e "s/CPU/rocket/" -e "s/MEMSIZE/0x8000000/" -e "s/BOARD/nexys4_ddr/" ../generic.dts | dtc -I dts -O dtb -o nexys4_ddr_rocket.dtb
<stdin>:87.45-90.19: Warning (unique_unit_address): /soc/lowrisc-gpio@44000000: duplicate unit-address (also used in node /soc/lowrisc-rtc@44000000)
echo -e .section .dtb \; .align 4,0 \; .globl _dtb \; _dtb: \; .incbin \"nexys4_ddr_rocket.dtb\" | riscv64-buildroot-linux-gnu-gcc -Os -ggdb -march=rv64imafdc -mabi=lp64d -Wall -mcmodel=medany -fno-builtin-printf -fno-builtin-memcpy -fno-builtin -ffreestanding -fshort-wchar -DBITSIZE=0x400000  -c -xassembler - -o nexys4_ddr_rocket.o
{standard input}: Assembler messages:
{standard input}:1: Error: junk at end of line, first unrecognized character is `-'
make[2]: *** [nexys4_ddr_rocket.o] Error 1
Makefile:41: recipe for target 'nexys4_ddr_rocket.o' failed
make[2]: Leaving directory '/home/leo/makess/lowrisc-chip-ariane-v0.7/fpga/src/etherboot'
make[1]: *** [fpga/src/etherboot/nexys4_ddr_rocket.sv] Error 2
Makefile:41: recipe for target 'fpga/src/etherboot/nexys4_ddr_rocket.sv' failed
make[1]: Leaving directory '/home/leo/makess/lowrisc-chip-ariane-v0.7'

It seems that the file: home/leo/makess/lowrisc-chip-ariane-v0.7/fpga/src/etherboot/Makefile has some mistakes around the -xassembler - -o, but I don't know how to fix it.

I also found that the 0.7rc2 repo structure seems very different from the 0.6 one, could you please explain a little bit about it?

Another issue: the binary releases of 0.7rc2 have a problem of don't have an ethernet connection(I step-by-step followed the quickstart guide). While booting, debian says: FAILED TO RAISE NETWORK INTERFACE. And this not just happened to me, my other friend has the same problem, I think it is a general issue. 0.6 doesn't have this issue.

The full (Not that full cuz it was too long) command line echos are in the attachment file.
[command line echos.txt](https://github.com/lowRISC/lowrisc-chip/files/4840616/command.line.echos.txt)

jrrk, 2020-06-27:
Hi Leo,
  Thanks for trying LowRISC. The first error is probably because/bin/sh is aliased to dash instead of bash. The second error is more mysterious and may be because a different version of device tree compiler than the one expected is installed. This could be due to running on an OS different to the supported one. To debug, run the assembly stage manually instead of in a pipe.

The third issue with Debian Ethernet, I am not clear if you do not want Ethernet or you have it and it doesn’t work. You can either use buildroot as recommended in the instructions, or chroot into the startup disk on your workstation to disable unwanted services.

Regarding the file layout, this closely follows the Ariane structure which was introduced in this release. It is supposed to reduce the number of independent stages needing to be run.

LeooLevinn, 2020-06-27:
Hi dev
   Thanks for your patient reply, I shall try to debug following your intro.
    
   About the third issue: Excuse me for didn't explain it clearly. I mean I do want the ethernet, but it does not work on both my nexys4ddr board and my friend's n4ddr board. Firstly, this issue only happens on 0.7rc2. We followed the quickstart intro, and it successfully booted Debian. But while Debian is booting, the error occurred: FAILD TO RAISE NETWORK INTERFACE, the full startup echos are as follows:

> Xshell 6 (Build 0197)
Copyright (c) 2002 NetSarang Computer, Inc. All rights reserved.

Type `help' to learn how to use Xshell prompt.
[C:\~]$ 

Connecting to COM4...
Connected.

Hello World!
Setup MAC addr
              SD boot
                         sd_f:  0x0  dma_f: 0x0 status: 
                                                            resp_vec: 0x0  init_state: 0xa  counter: 0x573
                                                                                                              init_fsm: 0x1  tran_state: 0x3f  tran_fsm: 0x0
                                                                                                                                                                sd_f:  0x0  dma_f: 0x0 status: 
                                                                                                                                                                                                   resp_vec: 0x0  init_state: 0xa  counter: 0x7260a
                                    init_fsm: 0x1  tran_state: 0x3f  tran_fsm: 0x0
                                                                                      sd_f:  0x0  dma_f: 0x0 status: 
                                                                                                                         resp_vec: 0x0  init_state: 0x14  counter: 0x0
                                                                                                                                                                          init_fsm: 0x0  tran_state: 0x3f  tran_fsm: 0x0
         sd_f:  0x0  dma_f: 0x0 status: 
                                            resp_vec: 0x0  init_state: 0x47  counter: 0x3ab68
                                                                                                 init_fsm: 0x1  tran_state: 0x3f  tran_fsm: 0x0
                                                                                                                                                   sd_f:  0x0  dma_f: 0x0 status: 
                                                                                                                                                                                      resp_vec: 0x0  init_state: 0x31  counter: 0x0
                    init_fsm: 0x0  tran_state: 0x3f  tran_fsm: 0x0
                                                                      sd_f:  0x0  dma_f: 0x0 status:  REQ_RDY INIT_DONE HCXC
                                                                                                                                resp_vec: 0x0  init_state: 0xf0  counter: 0x0
                                                                                                                                                                                 init_fsm: 0x0  tran_state: 0x0  tran_fsm: 0x0
               sd_f:  0x0  dma_f: 0x0 status:  REQ_RDY INIT_DONE HCXC
                                                                         resp_vec: 0x0  init_state: 0xf0  counter: 0x0
                                                                                                                          init_fsm: 0x0  tran_state: 0x0  tran_fsm: 0x0
                                                                                                                                                                       Load boot.bin into memory
                                                                                                                                                                                                load elf to DDR memory
   Section[0]: elfn(80000000,0x1000,0x8e80);
                                            Section[1]: elfn(80009000,0xa000,0x10c0);
                                                                                     memset(8000a0c0,0,0x9fc0);
                                                                                                               Section[2]: elfn(80200000,0x200000,0x754f3c);
                                                                                                                                                            Boot the loaded program at address $pc=800000bbl loader
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 5.3.18 (jrrk2@brexit.cl.cam.ac.uk) (gcc version 9.2.0 (Buildroot 2019.11.1-00008-g37e845d3fb)) #1 Mon Feb 17 15:51:51 GMT 2020
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] On node 0 totalpages: 32256
[    0.000000]   DMA32 zone: 504 pages used for memmap
[... cut ...] e is as follows):
[image]
But the rootfs.tar.xz of lowrisc 0.7 rc2 is beyond 2G after extraction, so 2G is not enough for 0.7rc2 (but it's enough for 0.6), the makefile needs a little modify. I give 4G for the Linux partition using the following partition table:
[image]
I believe that it will be better to correct the Makefile in order to avoid 'No space left on device' error, this will be more friendly to beginners.
    (Sorry for my poor English, I'm not a native English speaker...)

Thx again!

jrrk2, 2020-06-27:
Hi Leo,
  The ariane project which the v0.7 release is based on uses the non-project mode of Vivado, so you can't open the GUI to use the usual analysis functions. However some of the functionality is available by opening a checkpoint file in Vivado which is one of the outputs generated during the flow. Well done for spotting that partition overflow, I must admit I haven't really used the Debian image recently and it could have grown in the meantime.

I'm sorry to say I have left the project now, and as the rest of the team have previously migrated to work on OpenTitan, there is not likely to be an RC3 or full release with Debian. Meanwhile the RISCV support in Debian is evolving and this bug could be fixed if you follow the instructions to build Debian from the upstream installer scripts. I get the impression that you are using the canned Debian image from our website at the moment. This procedure has some problems and I don't know if they have been fixed yet. You are probably aware that the RISCV support failed to reach the QA threshold to include in mainline Debian at the Buster release, mainly due to a lack of rust support.

In view of all the above, the decision was taken to concentrate mainly on buildroot support for this release. As far as I know the Ethernet support is working well with the included buildroot image. If not, you can post a follow up. Debian includes a variety of tools to debug network problems. One help would be to connect your board directly to a laptop and setup a static Ethernet connection, using wireshark to monitor the connection for any packet problems. I think you will be able to learn a lot that way and maybe even fix the underlying problem yourself.

Another option to consider is the Fedora rawhide image for RISCV, I cannot go into details though, it is complicated.

LeooLevinn, 2020-06-28:
Hi dev
I'm surprised to know that you have left the project because you are still giving much help in the issue comments.  I could say nothing else but thanks, for your patience and specialized answer. 
Now I'm trying to use buildroot method seeing whether this error disappears. If that works, maybe I could post my bitstreams and Debian image on Github in order to let more beginners have a better experience with LowRISC.

LeooLevinn, 2020-06-28:
Hi Dev
I'm sorry to bother again, but I want to know what is the gcc and g++ version you used in your developing environment. The second 'mysterious' issue may be caused by incompatible gcc\g++ versions according to my investigation. I'm running Ubuntu18.04LTS,gcc\g++ 7.5.0 ,Vivado2018.2.

jrrk, 2020-06-28:
The gcc/g++ I was using was the one that comes with Ubuntu18.04LTS. I was not aware that there was anything special about it. It might be difficult to get access to that exact system again.

jrrk2, 2020-06-28:
Just to clarify the above, I assumed you were asking about the native gcc for some reason. The riscv5 gcc that I am using is the one produced by the buildroot environment path, buildroot-2019.11.1-lowrisc/rescuefs/host/bin/riscv64-buildroot-linux-gnu-gcc relative to the repo root. This is more likely to be relevant to whether your firmware build works or not. This is gcc version 9.2.0 (Buildroot 2019.11.1-07125-g37e845d). You can use 'make firmware' as a quick way to execute this step.

`

make -B -C fpga/src/etherboot BOARD=nexys4_ddr BITSIZE=0x400000 XILINX_PART=xc7a100tcsg324-1 XILINX_BOARD=digilentinc.com:nexys4_ddr:part0:1.1 CPU=rocket VENDOR=sifive MEMSIZE=0x8000000 CLK_PERIOD_NS="20" PATH=/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin:/usr/local/bin:/usr/bin:/bin
make[1]: Entering directory '/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7/fpga/src/etherboot'
CC    <= src/dhcp-client.c
CC    <= src/elf.c
CC    <= src/ethlib.c
CC    <= src/ff.c
CC    <= src/hash-md5.c
CC    <= src/mini-printf.c
CC    <= src/random.c
CC    <= src/string.c
CC    <= src/tftp_c.c
CC    <= src/uart.c
CC    <= src/boot.c
CC    <= src/dram.c
CC    <= src/cachetest.c
CC    <= src/hid.c
CC    <= src/logo.c
CC    <= src/queryTimeServer.c
CC    <= src/bt.c
CC    <= startup.S
sed -e "s/VENDOR/sifive/" -e "s/CPU/rocket/" -e "s/MEMSIZE/0x8000000/" -e "s/BOARD/nexys4_ddr/" ../generic.dts | dtc -I dts -O dtb -o nexys4_ddr_rocket.dtb
<stdin>:87.45-90.19: Warning (unique_unit_address): /soc/lowrisc-gpio@44000000: duplicate unit-address (also used in node /soc/lowrisc-rtc@44000000)
echo -e .section .dtb \; .align 4,0 \; .globl _dtb \; _dtb: \; .incbin \"nexys4_ddr_rocket.dtb\" | riscv64-buildroot-linux-gnu-gcc -Os -ggdb -march=rv64imafdc -mabi=lp64d -Wall -mcmodel=medany -fno-builtin-printf -fno-builtin-memcpy -fno-builtin -ffreestanding -fshort-wchar -DBITSIZE=0x400000  -c -xassembler - -o nexys4_ddr_rocket.o
riscv64-buildroot-linux-gnu-gcc -Os -ggdb -march=rv64imafdc -mabi=lp64d -Wall -mcmodel=medany -fno-builtin-printf -fno-builtin-memcpy -fno-builtin -ffreestanding -fshort-wchar -DBITSIZE=0x400000  -nostdlib -nodefaultlibs -nostartfiles -Wl,-Map=mapfile -I./ -I./src -Tlinker.lds startup.o nexys4_ddr_rocket.o src/dhcp-client.o src/elf.o src/ethlib.o src/ff.o src/hash-md5.o src/mini-printf.o src/random.o src/string.o src/tftp_c.o src/uart.o src/boot.o src/dram.o src/cachetest.o src/hid.o src/logo.o src/queryTimeServer.o src/bt.o src/main.c -o nexys4_ddr_rocket.elf
LD    >= nexys4_ddr_rocket.elf
riscv64-buildroot-linux-gnu-objcopy -O binary nexys4_ddr_rocket.elf nexys4_ddr_rocket.bin
dd if=nexys4_ddr_rocket.bin of=nexys4_ddr_rocket.img bs=128
250+1 records in
250+1 records out
32005 bytes (32 kB, 31 KiB) copied, 0.0011362 s, 28.2 MB/s
riscv64-buildroot-linux-gnu-objdump -d -l -S nexys4_ddr_rocket.elf > nexys4_ddr_rocket.dis
iverilog ../../scripts/cnvmem.v -o cnvmem
riscv64-buildroot-linux-gnu-objcopy -O verilog nexys4_ddr_rocket.elf nexys4_ddr_rocket.vlog
./cnvmem +VLOG=nexys4_ddr_rocket.vlog +MEM=nexys4_ddr_rocket.mem +SV=nexys4_ddr_rocket.sv
First = 87fe0000, Last = 87fe7fff
./cnvmem +VLOG=nexys4_ddr_rocket.vlog +MEM=nexys4_ddr_rocket.mem +SV=nexys4_ddr_rocket.sv
First = 87fe0000, Last = 87fe7fff
zero stage etherboot loader has been compiled!
make[1]: Leaving directory '/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7/fpga/src/etherboot'
make -B -C fpga/src/etherboot BOARD=nexys4_ddr BITSIZE=0x400000 XILINX_PART=xc7a100tcsg324-1 XILINX_BOARD=digilentinc.com:nexys4_ddr:part0:1.1 CPU=ariane VENDOR=ethz MEMSIZE=0x8000000 CLK_PERIOD_NS="20" PATH=/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin:/usr/local/bin:/usr/bin:/bin
make[1]: Entering directory '/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7/fpga/src/etherboot'
CC    <= src/dhcp-client.c
CC    <= src/elf.c
CC    <= src/ethlib.c
CC    <= src/ff.c
CC    <= src/hash-md5.c
CC    <= src/mini-printf.c
CC    <= src/random.c
CC    <= src/string.c
CC    <= src/tftp_c.c
CC    <= src/uart.c
CC    <= src/boot.c
CC    <= src/dram.c
CC    <= src/cachetest.c
CC    <= src/hid.c
CC    <= src/logo.c
CC    <= src/queryTimeServer.c
CC    <= src/bt.c
CC    <= startup.S
sed -e "s/VENDOR/ethz/" -e "s/CPU/ariane/" -e "s/MEMSIZE/0x8000000/" -e "s/BOARD/nexys4_ddr/" ../generic.dts | dtc -I dts -O dtb -o nexys4_ddr_ariane.dtb
<stdin>:87.45-90.19: Warning (unique_unit_address): /soc/lowrisc-gpio@44000000: duplicate unit-address (also used in node /soc/lowrisc-rtc@44000000)
echo -e .section .dtb \; .align 4,0 \; .globl _dtb \; _dtb: \; .incbin \"nexys4_ddr_ariane.dtb\" | riscv64-buildroot-linux-gnu-gcc -Os -ggdb -march=rv64imafdc -mabi=lp64d -Wall -mcmodel=medany -fno-builtin-printf -fno-builtin-memcpy -fno-builtin -ffreestanding -fshort-wchar -DBITSIZE=0x400000  -c -xassembler - -o nexys4_ddr_ariane.o
riscv64-buildroot-linux-gnu-gcc -Os -ggdb -march=rv64imafdc -mabi=lp64d -Wall -mcmodel=medany -fno-builtin-printf -fno-builtin-memcpy -fno-builtin -ffreestanding -fshort-wchar -DBITSIZE=0x400000  -nostdlib -nodefaultlibs -nostartfiles -Wl,-Map=mapfile -I./ -I./src -Tlinker.lds startup.o nexys4_ddr_ariane.o src/dhcp-client.o src/elf.o src/ethlib.o src/ff.o src/hash-md5.o src/mini-printf.o src/random.o src/string.o src/tftp_c.o src/uart.o src/boot.o src/dram.o src/cachetest.o src/hid.o src/logo.o src/queryTimeServer.o src/bt.o src/main.c -o nexys4_ddr_ariane.elf
LD    >= nexys4_ddr_ariane.elf
[... cut ...]
riscv64-buildroot-linux-gnu-objcopy -O verilog nexys4_ddr_ariane.elf nexys4_ddr_ariane.vlog
./cnvmem +VLOG=nexys4_ddr_ariane.vlog +MEM=nexys4_ddr_ariane.mem +SV=nexys4_ddr_ariane.sv
First = 87fe0000, Last = 87fe7fff
./cnvmem +VLOG=nexys4_ddr_ariane.vlog +MEM=nexys4_ddr_ariane.mem +SV=nexys4_ddr_ariane.sv
First = 87fe0000, Last = 87fe7fff
zero stage etherboot loader has been compiled!
make[1]: Leaving directory '/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7/fpga/src/etherboot'
jonathan@jrrk2-iMac:/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7$ 
`

LeooLevinn, 2020-06-28:
Thanks a lot, I will continuously dealing with this mysterious behavior.

jrrk2, 2020-06-28:
If you have an older riscv gcc in the path I would remove it and the RISCV environment variable from the path before you start to build.

LeooLevinn, 2020-06-28:
Thx, I'm working on it.

LeooLevinn, 2020-06-30:
Dear jrrk
Thanks for your help, now I can run "make firmware" successfully. 
This is how I dealt with it:
1. For the "|&" issue, I changed all of the "|&" to " 2>&1 | ", then no more errors.
2. For the 'mysterious' error, I found it could result in the difference between dash and bash. Seems like dash cannot recognize the "echo -e" command. After switching to bash, problem solved.

closed 2020-06-30 by LeooLevinn

LeooLevinn, 2020-07-01:
Dear jrrk
Just to report another progress, I successfully fixed the ethernet failure issue on Lowrisc 0.7rc2 prebuild (quickstart) demo. Maybe these contents could be added to the doc so others can solve this also.

It seems like the error 'Failed to raise network interfaces' is caused by the lack of a DHCP client. So I tried the following steps, and it works. 

1. Download these three packages from: https://packages.debian.org/sid/riscv64/allpackages
(needed to be manually installed in the following steps):

> libisc-export1105_9.11.19+dfsg-1_riscv64.deb
> libdns-export1110_9.11.19+dfsg-1_riscv64.deb
> isc-dhcp-client_4.4.1-2.1+b2_riscv64.deb

2. Make sure the SD-card can boot 0.7rc2 prebuild (quickstart) Debian system. Then plug the sd-card into a PC (or VM) running Linux, such as Ubuntu.18.04 LTS

3. Enter the system dictionary of lowrisc-Debian, copy the above three .deb packages into a dictionary that you can remember (such as /root)

4. Plug the SD-Card back to N4DDR board, open it up, login to the system.

5. Install the above packages using `dpkg -i` command. Firstly install the libisc, then libdns, then isc-dhcp-client (maybe firstly libdns then libisc then isc-dhcp-client? I can't remember). If the installing sequence of the three packages was wrong, a dependency error will show up.

6. `sudo shutdown -h now`. After safely shutdown, reboot it, and the network should work fine (try ping command). 

Tips: make sure to `use shutdown -h now` command on 0.7, don't just turn off the power switch, or you will be unable to load Debian next time. (However, force power-off is fine on 0.6)

jrrk, 2020-07-01:
The default startup daemon is systemd which is meant to include DHCP internally from what I understand. Nevertheless this is good detective work, but all your customisations could be done as part of the Debian image preparation (and there is no need to do it live) as part of the bootstrap process, or later automatic customisations.