lowRISC/lowrisc-chip #176: some questions about ariane-v0.7

michaelwvw, 2020-07-13:
I'm using nexys4-ddr-rocket.
1. There is no apt-get available in the embeded linux, then how can i install a software. Just for study, so the performance is not an issue.
2. I cannot find the GPIO device in this embeded linux, then how to control GPIO/LED in linux? I can use the GPIO device in refresh-v0.6.
3. There is no embeded version gcc which is available in refresh-v0.6. Then i tried to compile program in pc linux and then execute it in embeded linux, but it's reporting "./hello: /lib64/libc.so.6: version `GLIBC_2.26' not found (required by ./hello)". How to resolve this issue? Or is there an alternative way to compile a program for embeded linux?

jrrk, 2020-07-13:
The distribution was slimmed down a lot to make room for X-windows. You can still install Debian as an option if you are an ‘advanced user’. See below :

> 
> 
> I'm using nexys4-ddr-rocket.
> 
> There is no apt-get available in the embeded linux, then how can i install a software. Just for study, so the performance is not an issue.
Apt is Debian specific. By default buildroot is installed because Debian was a bit unstable at the time. The recommended route is to add your software as local buildroot packages.
> I cannot find the GPIO device in this embeded linux, then how to control GPIO/LED in linux? I can use the GPIO device in refresh-v0.6.
This function has not been ported from the earlier release. It should be straightforward for someone else to do however.
> There is no embeded version gcc which is available in refresh-v0.6. Then i tried to compile program in pc linux and then execute it in embeded linux, but it's reporting "./hello: /lib64/libc.so.6: [...]
The appropriate cross compiler is installed within buildroot. It has not been made available native for some reason.

michaelwvw, 2020-07-13:
Thanks jrrk!
But looks like i have to read buildroot docs before i understand your comments.

michaelwvw, 2020-07-17:
Hi, Jrrk,
I'm trying to add GPIO function into BR2, then which one should be enabled here. Suppose it's GPIO_XILINX while not the GPIO used in v0.6, am'i right?
I also see "GPIO Bitbanged LowRISC" is enabled, what's it for?

michaelwvw, 2020-07-17:
Sorry, missed one thing, since there are address overlay, should disable RTC if enable GPIO?

jrrk, 2020-07-17:
These functions just enable a pre-processor symbol, they have no effect unless the appropriate kernel driver is loaded. There is a generic Linux GPIO, that would be the proper way to go I think.

jrrk, 2020-07-17:
Both functions may be handled by a single driver.