lowRISC/lowrisc-chip #177: How to add new peripheral to Refresh 0.6 SoC

Ankabs, 2020-08-27:
I am trying to add new peripheral to SoC to Refresh 0.6, I created a vivado project and examined codes. I think right place to add a peripheral is perip_soc.sv file. But i cannot see axi interface signals or addresses for peripherals. Is there something i am missing?

jrrk, 2020-08-27:
This release lacks an AXI hub so it is more difficult to add peripherals. The address map will depend on how you wire up the address lines after the AXI2MEM converter.

Ankabs, 2020-08-27:
I choose 0.6 version because i can compile and run a custom C program on LowRISC v0.6. Also in the end I should connect my custom IP to SoC and drive it using my custom software. In issue named as "some questions about ariane-v0.7 #176" you stated that debian is unstable to date. To accomplish my tasks (dive custom peripherals using custom software on Debian) what is the recommended route? Is there a version in which doing these tasks is easier?

jrrk, 2020-08-27:
Both v0.6 and v0.7 have a big investment in them. The Ariane infrastructure is oriented to adding new peripherals and certainly has better disk performance as well as X-Windows. The use of buildroot is not to everybody’s taste though, but it is more stable than Debian unstable.

hgurbuzz, 2022-02-19:
Hi. I want to make an iot application in LowRISC v0.6. I saw that there is no peripheral device that I can use for this application other than ethernet. Am I wrong? Which peripherals are supported.

jrrk2, 2022-02-20:
This release was designed for a low cost board that only has Ethernet. You can support other standards such as Bluetooth using a PMOD and the v0.7 pre-release, or with your own custom hardware, but this should only be attempted by an expert.