lowRISC/lowrisc-chip #180: Segmentation Error

HanningChen, 2020-10-16:
Hi, I am working on lowrisc-chip KC705 repository. I faced an exact issue as issue#91: 
https://github.com/lowRISC/lowrisc-chip/issues/91
However, based on issue#91, I only know that my environment variable $LD_LIBRARY_PATH is wrong but I don't know how to fix this?

 Here is the result of "echo $LD_LIBRARY_PATH":
/home/hanning/2020Fall/kc705_update/tools/lib:/home/hanning/2020Fall/kc705_update/tools/lib:/home/hanning/2020Fall/kc705_update/tools/lib:/home/hanning/Xilinx/Vivado/2015.4/lib/lnx64.o:/opt/ros/dashing/opt/yaml_cpp_vendor/lib:/opt/ros/dashing/opt/rviz_ogre_vendor/lib:/opt/ros/dashing/lib:/home/hanning/2020Fall/kc705_update/riscv/lib:/home/hanning/2020Fall/kc705_update/riscv/lib:/home/hanning/2020Fall/kc705_update/riscv/lib

I am not sure which part is wrong or I miss setting some variable.

The repository I am working on is: lowRISC/lowrisc-chip, branch: kc705_update
My Vivado version is: 2015.04 and my PC os is: ubuntu 18.04

Thanks

jrrk, 2020-10-16:
When you source the environment for Vivado, it will point LD_LIBRARY_PATH to a version of shared libraries which is suitable for 2015.4, but not suitable for the system which is much newer. Try unset LD_LIBRARY_PATH and regenerate any invalid files (typically zero sized files)

HanningChen, 2020-10-16:
Thanks for help, I will try this tonight.

closed 2020-12-02 by HanningChen