lowRISC/lowrisc-chip #183: Binary Release Documentation Question

johnduffymsc, 2020-11-20:
Hi

I'm trying to install the lowRISC SoC on a Nexys 4 DDR board, and I'm a bit confused by the references to SD cards and USB cards in the _Getting Started with the refresh-v0.6 prebuilt binaries_ documentation.

Does the _make memstick USB=sdc_ command refer to the actual USB memory stick used to hold the bit stream, whereas all previous make commands, e.g. _make USB=sdc cleandisk partition_, refer to an SD card in a USB adaptor?

Kind regards

John

jrrk, 2020-11-20:
make memstick is indeed for use with the USB bitstream config option. The RISCV itself cannot see this disk. It is not much used because it precludes the use of the keyboard.

johnduffymsc, 2020-11-20:
Thanks, I've literally just figured it out. I have the root file system on the SD card, and the bit stream on the USB stick, and I can ping a booted _lowrisc_.