lowRISC/lowrisc-chip #184: Unrecognized character error while building nexys4_ddr_ariane.o

abdullahyildiz, 2020-11-21:
I'm trying to build FPGA bitstream of lowRISC with Ariane. I'm working on [ariane-v0.7](https://github.com/lowRISC/lowrisc-chip/tree/ariane-v0.7) branch.

I get the following error when riscv64-buildroot-linux-gnu-gcc tries to read from pipe:

```
$ make nexys4_ddr_ariane
...
make fpga/work-fpga/nexys4_ddr_ariane/ariane_xilinx.mcs BOARD="nexys4_ddr" CPU="ariane" BITSIZE=0x400000 XILINX_PART="xc7a100tcsg324-1" XILINX_BOARD="digilentinc.com:nexys4_ddr:part0:1.1" VENDOR="ethz" MEMSIZE="0x8000000" BBL=/home/edaman/lowrisc-chip-ariane-v0.7/lowrisc-quickstart/rescue.bin 2>&1 | tee nexys4_ddr_ariane.log
make[1]: Entering directory '/home/edaman/lowrisc-chip-ariane-v0.7'
make -C fpga/src/etherboot BOARD=nexys4_ddr CPU=ariane VENDOR=ethz MEMSIZE=0x8000000 PATH=/home/edaman/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin:/usr/local/bin:/usr/bin:/bin
make[2]: Entering directory '/home/edaman/lowrisc-chip-ariane-v0.7/fpga/src/etherboot'
echo -e .section .dtb \; .align 4,0 \; .globl _dtb \; _dtb: \; .incbin \"nexys4_ddr_ariane.dtb\" | riscv64-buildroot-linux-gnu-gcc -Os -ggdb -march=rv64imafdc -mabi=lp64d -Wall -mcmodel=medany -fno-builtin-printf -fno-builtin-memcpy -fno-builtin -ffreestanding -fshort-wchar -DBITSIZE=0x400000  -c -xassembler - -o nexys4_ddr_ariane.o
{standard input}: Assembler messages:
{standard input}:1: Error: junk at end of line, first unrecognized character is `-'
Makefile:41: recipe for target 'nexys4_ddr_ariane.o' failed
make[2]: *** [nexys4_ddr_ariane.o] Error 1
make[2]: Leaving directory '/home/edaman/lowrisc-chip-ariane-v0.7/fpga/src/etherboot'
Makefile:41: recipe for target 'fpga/src/etherboot/nexys4_ddr_ariane.sv' failed
make[1]: *** [fpga/src/etherboot/nexys4_ddr_ariane.sv] Error 2
make[1]: Leaving directory '/home/edaman/lowrisc-chip-ariane-v0.7'
```

Is this related to shell type/version?

jrrk, 2020-11-21:
Indeed the flow requires that /bin/sh is aliased to /bin/bash (not the default). But I cannot recall if this relates to your problem.

abdullahyildiz, 2020-11-21:
> Indeed the flow requires that /bin/sh is aliased to /bin/bash (not the default). But I cannot recall if this relates to your problem.

Yes, you are right. I forgot to change the default shell from dash to bash.

closed 2020-11-21 by abdullahyildiz