lowRISC/lowrisc-chip #185: Error with 'make program-updated'

kidonglee, 2020-11-23:
Hi, 

I am trying the sequences in https://www.lowrisc.org/docs/.
    make bitstream  // -> OK
    make cfgmem  // -> OK
    make program-cfgmem // -> OK
    make program-updated // -> Error
With the command 'make program-updated', I got error messages as follow,
```
data2mem -bm src/boot.mem -bd src/boot.mem -bt lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.bit -o b lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.new.bit
ERROR: Could not find 64-bit executable.
ERROR: /opt/Xilinx/Vitis/2020.1/bin/unwrapped/lnx64.o/data2mem does not exist
Makefile:204: recipe for target 'lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.new.bit' failed
make: *** [lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.new.bit] Error 1
```
Please help me.

Thank you.

kidonglee, 2020-11-23:
Hi, 

I found that data2mem is deprecated in Vitis.
And there was no problem with XSDK.
BTW, is there any plan to support Vitis ?

Thanks.

jrrk, 2020-11-23:
The latest version of Vivado that is supported is 2018.x . You can use a newer version but you will not have the ability to update the ROM inside the compiled bitstream. The vast majority of functionality can work without changing the ROM however.

kidonglee, 2020-11-26:
Dear, jrrk
Thank you for the kind explanation.

closed 2020-11-26 by kidonglee