lowRISC/lowrisc-chip #186: Remote Installation Dumb Question

johnduffymsc, 2020-11-25:
Apologies for this probably dumb question…

I have installed lowRISC on my Nexys 4 DDR board using a USB memory stick to hold the bit file. I can boot and ping the “lowrisk” machine, and ssh works but I (obviously) can’t login as root.

How do I login using the serial installation method? I have the USB cable that comes with the Nexys board, and I can see that it fits into the mini USB/uart socket, but I’m a bit stumped how to use a terminal emulator with a USB connector.

Any help would be much appreciated.

jrrk, 2020-11-25:
The serial port console appears as /dev/ttyUSB? On your host PC. You can use microcom on Linux or putty on Windows. It expects to talk at 115200 baud and root logon is allowed.

johnduffymsc, 2020-12-02:
Thanks. After plugging in the USB/serial cable into my Pi 4 and Nexys board, and then powering the board up, /dev/ttyUSB0 appeared. I created a new user and I can now ssh into lowrisc.

Many thanks

closed 2020-12-02 by johnduffymsc