lowRISC/lowrisc-chip #187: Timing constraints are not met in ariane-v0.7

abdullahyildiz, 2020-11-25:
I encountered many timing violations that are related to `ui_clk` during the synthesis of Ariane based SoC on Nexys 4 DDR board. I noticed that the core subsystem takes `ui_clk` that is generated by `xlnx_mig_7_ddr_nexys4_ddr` IP and `ui_clk` [must be a quarter the frequency of the DRAM clock](https://www.xilinx.com/support/documentation/ip_documentation/ug586_7Series_MIS.pdf).

Is there anyone that encountered this issue before?

jrrk, 2020-11-25:
The Nexys4ddr chip is a tight squeeze for Ariane, nevertheless we felt it was important to offer continuity of board support from the previous release. The Genesys2 board uses a faster technology and if I recall correctly does meet timing.

In spite of this I have seen demonstrable results on Nexys.

If your question is about how to slow down the DDR clock to ensure timing is always met, you need to examine the GUI configuration for the DDR IP for a different valid value, or use an AXI clock up-converter. This is outside the scope of this support forum I’m afraid.

abdullahyildiz, 2020-11-25:
@jrrk,

I haven't tested the design on the board yet.

I tried to decrease the current DDR clock frequency however GUI for MIG 7 series restricts the selection of clock period to between 3000ps and 5000ps.

closed 2020-11-25 by abdullahyildiz