lowRISC/lowrisc-chip #188: loading bitstream file into FPGA but no reaction happened on Tera Term

HanningChen, 2020-12-02:
Hi, I am working on the hello world test of "lowrisc-chip". The branch that I am working on is kc705_update. The problem is that I can generate the bitstream file and combine it with the "hello world" test hex file, but when I load it into my FPGA, there is no reaction on the terminal.

The board that I am using is Xilinx Atrix-7 AC701 board. I made a lot of modifications based on the "kc705_update" branch including pin mapping, MIG, BRAM controller, etc. The whole synthesis and implementation look normal. Since I don't have Xilinx Kintex KC705 board, I am not sure whether this "kc705_update" is correct or not. My final target is to build a RISC-V CPU on Xilinx AC701 board and port Linux on it, so I choose KC705 board project as a base.

For now, after checking Xilinx official BIST (Build In Self Test), I guess the problem is the test .c file is wrong. But I am not sure. Thanks for your help.

jrrk, 2020-12-02:
I haven’t investigated the AC701 board at all. The kc705_update branch was created explicitly for the KC705 and has no relevance to other boards (except as an exemplar of what needs to be changed. If nothing works at all the most likely reason is that your frequency divider is setup wrong for that board. Unfortunately support for porting to new boards is a lengthy task and outside the scope of mailing list support that can be realistically be offered, sorry about that.

HanningChen, 2020-12-03:
Thank you so much for your help. Could you explain a little more about why the frequency divider is the critical problem? I definitely change the clock_generator parameter in my design.
Also, it is fine it is not supported. I guess port one design from one board to another is challenging but interesting. I really learned a lot from this process. I just want some general idea of what might be the trap that I missed.

jrrk, 2020-12-03:
The main problem is the DDR interface, there will be a limited choice of frequencies supported by MIG, this will feed into the AXI controller to DDR ratio, and consequently the RISCV clock rate. You can add an asynchronous AXI frequency upscaler but the configuration can be a bit tricky, also the RISCV clock rate directly determines the UART clock rate. An additional problem is that the Vivado simulator isn’t adequate and you need an external simulator which is complicated to configure and needs an expensive license (unless you are a research institution).

HanningChen, 2020-12-03:
Thank you so much. I will think of it.