lowRISC/lowrisc-chip #191: Syntax error during make nexys4_ddr_rocket

pcotret, 2021-03-02:
Followed the instructions available at: https://www.cl.cam.ac.uk/~jrrk2/docs/docs/
```bash
make nexys4_ddr_rocket
```
I got the following error:
```bash
make fpga/work-fpga/nexys4_ddr_rocket/rocket_xilinx.mcs BOARD="nexys4_ddr" CPU="rocket" BITSIZE=0x400000 XILINX_PART="xc7a100tcsg324-1" XILINX_BOARD="digilentinc.com:nexys4_ddr:part0:1.1" VENDOR="sifive" MEMSIZE="0x8000000" BBL=/home/pascal/Documents/ensta/work/lowrisc/lowrisc-chip-ariane-v0.7/lowrisc-quickstart/rescue.bin |& tee nexys4_ddr_rocket.log
/bin/sh: 1: Syntax error: "&" unexpected
make: *** [Makefile:61: nexys4_ddr_rocket] Error 2
```

jrrk, 2021-03-02:
As explained in the docs, /bin/sh should be linked to bash instead of dash.

pcotret, 2021-03-03:
Got it, thanks.

closed 2021-10-04 by pcotret