lowRISC/lowrisc-chip #193: make error

kidonglee, 2021-04-23:
Dear Guys,

I am following the sequences in the page "https://www.cl.cam.ac.uk/~jrrk2/docs/docs/".
But I have a problem with the step of "Produce an FPGA Bitstream".
I have an error messages while building FPGA bitstream, as below.
With 'make nexys4_ddr_rocket', I also have the same message.
Are there any one who knows the solution on this problem?
Please help.
Thank you in advance.

Best Regards,

`
$> make nexys4_ddr_ariane
make -C buildroot-2019.11.1-lowrisc/mainfs
make[1]: Entering directory '/mnt/hdd/work/Wukong/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs'
>>> host-openocd f93ede5401c711e55d9852986aa399c0318efb22 Downloading
Reinitialized existing Git repository in /mnt/hdd/work/Wukong/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/dl/openocd/git/.git/
Fetching all references
Could not fetch special ref 'f93ede5401c711e55d9852986aa399c0318efb22'; assuming it is not special.
warning: refname 'f93ede5401c711e55d9852986aa399c0318efb22' is ambiguous.
Git normally never creates a ref that ends with 40 hex characters
because it will be ignored when you just specify 40-hex. These refs
may be created by mistake. For example,

  git checkout -b $br $(git rev-parse ...)

where "$br" is somehow empty and a 40-hex ref is created. Please
examine these refs and maybe delete them. Turn this message off by
running "git config advice.objectNameWarning false"
ERROR: openocd-f93ede5401c711e55d9852986aa399c0318efb22.tar.gz has wrong sha1 hash:
ERROR: expected: 1b55952e9b51891f77de09ec123aa94ab3b64fe9
ERROR: got     : 15e8f5af6b0c4d8bfb78f207c423b2247e4840d2
ERROR: Incomplete download, or man-in-the-middle (MITM) attack
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/home/kdlee/.wget-hsts'. HSTS will be disabled.
--2021-04-22 23:19:22--  http://sources.buildroot.net/openocd/openocd-f93ede5401c711e55d9852986aa399c0318efb22.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 172.67.72.56, 104.26.0.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-04-22 23:19:22 ERROR 404: Not Found.

Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/home/kdlee/.wget-hsts'. HSTS will be disabled.
--2021-04-22 23:19:22--  http://sources.buildroot.net/openocd-f93ede5401c711e55d9852986aa399c0318efb22.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.0.37, 172.67.72.56, 104.26.1.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.0.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-04-22 23:19:22 ERROR 404: Not Found.

package/pkg-generic.mk:143: recipe for target '/mnt/hdd/work/Wukong/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-openocd-f93ede5401c711e55d9852986aa399c0318efb22/.stamp_downloaded' failed
make[2]: *** [/mnt/hdd/work/Wukong/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-openocd-f93ede5401c711e55d9852986aa399c0318efb22/.stamp_downloaded] Error 1
Makefile:23: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/mnt/hdd/work/Wukong/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs'
Makefile:175: recipe for target '/home/kdlee/work/Wukong/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar' failed
make: *** [/home/kdlee/work/Wukong/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar] Error 2
`

jrrk2, 2021-04-23:
It looks like you downloaded an incomplete repo, suggest you delete your work area and start again from scratch.

kidonglee, 2021-04-25:
> It looks like you downloaded an incomplete repo, suggest you delete your work area and start again from scratch.

Dear Jrrk2,
As I mentioned above, I follow the sequence of "https://www.cl.cam.ac.uk/~jrrk2/docs/docs/".
So, I cloned the branch 'lowrisc-chip-ariane-v0.7', as follow,
```
git clone -b ariane-v0.7 --recursive https://github.com/lowrisc/lowrisc-chip.git lowrisc-chip-ariane-v0.7
```
Do you mean that this branch is incomplete?
And could you please let me know how I download it?
Thank you very much.

Best Regards,

wsong83, 2021-04-26:
Use the right clone command does not guarantee a success clone.
One of the usual reasons could be a time-out disconnection before a full clone due to slow or unstable internet connection.
What you can try is going into the cloned repo and repeatedly updating all submodules if the clone failed.

jrrk2, 2021-04-26:
If you look at your log you can see that it failed inside the buildroot target. This target builds Linux and the userland from separately downloaded sources. This seems to be the step that failed and if it happens again with a clean tree then this is outside of LowRISC control. You should be able to find a pre-compiled root filing system in the release area of GitHub, but this usage is for users who are familiar with our previous releases and is outside the scope of support that is possible from this forum.