lowRISC/lowrisc-chip #194: Tilelink conformance level

Marc43, 2021-04-29:
Hi, 
I was trying to tell which conformance level of TileLink do you implement but I was not able. 

Can you tell me which one?

Thanks and sorry if it is a stupid question!

jrrk2, 2021-04-29:
LowRISC releases up to v0.4 used TileLink-1, v0.5 onwards used TileLink-2. This protocol is confined to inside the Rocket coreplex, all peripherals use AXI or a simple memory bus.

Marc43, 2021-04-29:
Thanks for answering, but, this is not the same as the conformance level, right?

You can implement TL-UL, TL-UH or TL-C, which one of those is implemented between the L1 and the L2?

jrrk2, 2021-04-29:
As I understand it TileLink-1 never worked properly with an L2 cache, so can’t really be described as conformant to anything. TileLink-2 removed the L2 cache from the open-source Rocket database, so again we do not have it in later releases. Once SiFive started the Rocket database was pretty much orphaned. I do not know if SiFive released an L2 cache and if so, what the license terms are.

closed 2021-04-30 by Marc43