lowRISC/lowrisc-chip #196: The Adept version recommended by the documents are too old to be able to download from the  https://reference.digilentinc.com/software/adept/start

nicolast0604, 2021-06-04:
Hi 
The Adept version recommended by the documents are too old to be able to download from the 
https://reference.digilentinc.com/software/adept/start

Is it workable to use the newer version to replace it?

Thanks a lot.
Nicolas

USB-JTAG installation
The default JTAG and configuration method for both the KC705 and the NEXYS4-DDR kits is the UART-JTAG cable. A driver is needed for Vivado to utilize this cable. We recommend to use the driver packages provided by Digilent: https://www.digilentinc.com/Products/Detail.cfm?Prod=ADEPT2. For 64-bit Ubuntu, please download and install the following two deb packages:

Adept 2.16.1 Runtime, X64 DEB
Adept 2.2.1 Utilities, X64 DEB

jrrk, 2021-06-04:
The driver you need is now incorporated into Vivado, follow the instructions on our website to install it if the sys Administator hasn’t done so already.

jrrk, 2021-06-04:
If you are working on an old version of LowRISC and Vivado, you can most likely use the latest adept install though we have not tried it.