lowRISC/lowrisc-chip #197: ERROR: sizeof(size_t) doesn't match GLIB_SIZEOF_SIZE_T.

nicolast0604, 2021-06-04:
Hi 
After ran the command on lowrisc-chip v0,7
make buildroot

I got the following error. Is there any solution to fix it?

BR
Nicolas 

ainfs/host/bin:/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/sbin:/home/nicolast0604/.cargo/bin:/home/nicolast0604/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/tools/verilator/4.104/bin:/tools/verible/v0.0-1213-g9e5c085/bin:/home/nicolast0604/.cargo/env" PKG_CONFIG="/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/lib/pkgconfig:/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/share/pkgconfig" AR="/usr/bin/ar" AS="/usr/bin/as" LD="/usr/bin/ld" NM="/usr/bin/nm" CC="/usr/bin/gcc" GCC="/usr/bin/gcc" CXX="/usr/bin/g++" CPP="/usr/bin/cpp" OBJCOPY="/usr/bin/objcopy" RANLIB="/usr/bin/ranlib" CPPFLAGS="-I/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/include" CFLAGS="-O2 -I/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/include" CXXFLAGS="-O2 -I/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/include" LDFLAGS="-L/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/lib -Wl,-rpath,/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/lib" INTLTOOL_PERL=/usr/bin/perl CPP="/usr/bin/gcc -E" ./configure --target-list="riscv64-linux-user" --prefix="/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host" --interp-prefix=/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/riscv64-buildroot-linux-gnu/sysroot --cc="/usr/bin/gcc" --host-cc="/usr/bin/gcc" --extra-cflags="-O2 -I/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/include" --extra-ldflags="-L/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/lib -Wl,-rpath,/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/lib" --python=/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin/python3 --disable-system --enable-linux-user --static --disable-virtfs --disable-libusb

ERROR: sizeof(size_t) doesn't match GLIB_SIZEOF_SIZE_T.
       You probably need to set PKG_CONFIG_LIBDIR
       to point to the right pkg-config files for your
       build target

package/pkg-generic.mk:228: recipe for target '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_configured' failed
make[2]: *** [/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_configured] Error 1
Makefile:23: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs'
Makefile:175: recipe for target '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar' failed
make: *** [/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar] Error 2

nicolast0604, 2021-06-04:
It seems fixed by the command
apt install libglib2.0-dev
but I soon have the other error as below:

unset TARGET_DIR; PATH="/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin:/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/sbin:/home/nicolast0604/.cargo/bin:/home/nicolast0604/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/tools/verilator/4.104/bin:/tools/verible/v0.0-1213-g9e5c085/bin:/home/nicolast0604/.cargo/env" PKG_CONFIG="/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/lib/pkgconfig:/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/host/share/pkgconfig" /usr/bin/make -j7 -C /home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1
	CHK version_gen.h
  LINK    qemu-nbd
/usr/bin/ld: cannot find -lxml2
collect2: error: ld returned 1 exit status
/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/rules.mak:124: recipe for target 'qemu-nbd' failed
make[3]: *** [qemu-nbd] Error 1
package/pkg-generic.mk:238: recipe for target '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_built' failed
make[2]: *** [/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_built] Error 2
Makefile:23: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs'
Makefile:175: recipe for target '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar' failed
make: *** [/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar] Error 2

jrrk, 2021-06-04:
Perhaps libxml2-dev is not installed