lowRISC/lowrisc-chip #198: /usr/bin/ld: cannot find -lxml2 collect2: error: ld returned 1 exit status

nicolast0604, 2021-06-05:
Error when ran the 
make buildroot

with v0.7

/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x267): warning: Using 'getpwuid' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
(.text+0xdd): warning: Using 'getpwnam_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
(.text+0x11b): warning: Using 'getpwuid_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
  LINK    ivshmem-server
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x267): warning: Using 'getpwuid' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
(.text+0xdd): warning: Using 'getpwnam_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
(.text+0x11b): warning: Using 'getpwuid_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
libqemuutil.a(qemu-sockets.o): In function `inet_parse_connect_saddr':
/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/util/qemu-sockets.c:406: warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
  LINK    qemu-nbd
/usr/bin/ld: cannot find -lxml2
collect2: error: ld returned 1 exit status
/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/rules.mak:124: recipe for target 'qemu-nbd' failed
make[3]: *** [qemu-nbd] Error 1
make[3]: *** Waiting for unfinished jobs....
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libglib-2.0.a(libglib_2_0_la-gutils.o): In function `g_get_user_database_entry':
(.text+0x267): warning: Using 'getpwuid' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
(.text+0xdd): warning: Using 'getpwnam_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
(.text+0x11b): warning: Using 'getpwuid_r' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
package/pkg-generic.mk:238: recipe for target '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_built' failed
make[2]: *** [/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/build/host-qemu-3.1.1.1/.stamp_built] Error 2
Makefile:23: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs'
Makefile:175: recipe for target '/home/nicolast0604/lowrisc-chip-ariane-v0.7/buildroot-2019.11.1-lowrisc/mainfs/images/rootfs.tar' failed

jrrk, 2021-06-05:
Did you try installing libxml2-dev as previously suggested?

nicolast0604, 2021-06-05:
Thanks a lot. It works after install  libxml2-dev