lowRISC/lowrisc-chip #201: ERROR: [Synth 8-6735] net type must be explicitly specified for 'clk_i' when default_nettype is none [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:26]

nicolast0604, 2021-06-06:
Ran make nexys4_ddr_rocket with Vivado 2018.3
and have ERRORs
any hint ?

WARNING: [filemgmt 56-315] Source scanning failed during design analysis. To get more details run synthesis or simulation and check the log.
# add_files -fileset constrs_1 -norecurse constraints/$::env(BOARD).xdc
# set_property include_dirs ../ariane/src/common_cells/include [current_fileset]
# synth_design -rtl -name rtl_1
Command: synth_design -rtl -name rtl_1
Starting synth_design
Using part: xc7a100tcsg324-1
WARNING: [Vivado_Tcl 4-393] The 'Synthesis' target of the following IPs are stale, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/xilinx/xlnx_mig_7_ddr_nexys4_ddr/nexys4_ddr/ip/xlnx_mig_7_ddr_nexys4_ddr.xci

WARNING: [Vivado_Tcl 4-393] The 'Implementation' target of the following IPs are stale, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/xilinx/xlnx_mig_7_ddr_nexys4_ddr/nexys4_ddr/ip/xlnx_mig_7_ddr_nexys4_ddr.xci

Top: rocket_xilinx
ERROR: [Synth 8-6735] net type must be explicitly specified for 'clk_i' when default_nettype is none [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:26]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'clk_200MHz_i' when default_nettype is none [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:27]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'rst_ni' when default_nettype is none [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:28]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'rx_i' when default_nettype is none [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:32]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'clk_rmii' when default_nettype is none [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:36]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'clk_rmii_quad' when default_nettype is none [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:37]
ERROR: [Synth 8-6735] net type must be explicitly specified for 'dip_switches_i' when default_nettype is none [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:69]
INFO: [Synth 8-2350] module ariane_peripherals_xilinx ignored due to previous errors [/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv:14]
Failed to read verilog '/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv'
1 Infos, 2 Warnings, 0 Critical Warnings and 8 Errors encountered.
synth_design failed
ERROR: [Vivado_Tcl 4-5] Elaboration failed - please see the console for details
INFO: [Common 17-206] Exiting Vivado at Sun Jun  6 22:53:01 2021...
Makefile:40: recipe for target 'work-fpga/nexys4_ddr_rocket/rocket_xilinx.bit' failed
make[2]: *** [work-fpga/nexys4_ddr_rocket/rocket_xilinx.bit] Error 1
make[2]: Leaving directory '/home/nicolast0604/lowrisc-chip-ariane-v0.7/fpga'
Makefile:51: recipe for target 'fpga/work-fpga/nexys4_ddr_rocket/rocket_xilinx.mcs' failed
make[1]: *** [fpga/work-fpga/nexys4_ddr_rocket/rocket_xilinx.mcs] Error 2
make[1]: Leaving directory '/home/nicolast0604/lowrisc-chip-ariane-v0.7'

jrrk, 2021-06-06:
Newer versions of Vivado are more strict, you can avoid this error by changing default_nettype none, to default_nettype wire (or use the Vivado version recommended in the documentation)

nicolast0604, 2021-06-07:
After back to 2018.2 it works.
For 2018.3, change default_nettype to wire makes make nexys4_ddr_ariane works but 

**make nexys4_ddr_rocket**
**fails** with the following error:**

Phase 2.1 Floorplanning
Phase 2.1 Floorplanning | Checksum: 142d260b8

Time (s): cpu = 00:00:21 ; elapsed = 00:01:26 . Memory (MB): peak = 3347.527 ; gain = 0.000 ; free physical = 199 ; free virtual = 333
/tools/Xilinx/Vivado/2018.3/bin/loader: line 213: 19353 Killed                  "$RDI_PROG" "$@"
Makefile:40: recipe for target 'work-fpga/nexys4_ddr_rocket/rocket_xilinx.bit' failed
make[2]: *** [work-fpga/nexys4_ddr_rocket/rocket_xilinx.bit] Error 137
make[2]: Leaving directory '/home/nicolast0604/lowrisc-chip-ariane-v0.7-2018.3/fpga'
Makefile:51: recipe for target 'fpga/work-fpga/nexys4_ddr_rocket/rocket_xilinx.mcs' failed
make[1]: *** [fpga/work-fpga/nexys4_ddr_rocket/rocket_xilinx.mcs] Error 2
make[1]: Leaving directory '/home/nicolast0604/lowrisc-chip-ariane-v0.7-2018.3'

nicolast0604, 2021-06-07:
It works either changes default_nettype wire on 2018.3 or use the 2018.2.
Thanks.