lowRISC/lowrisc-chip #204: Kernel panic with `0.7-rc2` release

pcotret, 2021-10-31:
Tried the Nexys4/Rocket files https://github.com/lowRISC/lowrisc-chip/releases/tag/v0.7-rc2

Linux seems to boot but I got a kernel panic:
```bash
[    5.474306] Run /etc/init as init process
[    5.478544] Run /bin/init as init process
[    5.482300] Run /bin/sh as init process
[    5.486184] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rs.
[    5.500674] CPU: 0 PID: 1 Comm: swapper Not tainted 5.3.18 #1
[    5.506494] Call Trace:
[    5.509216] [<ffffffe00002d1a4>] walk_stackframe+0x0/0xa6
[    5.514660] [<ffffffe00002d30a>] show_stack+0x2a/0x34
[    5.519842] [<ffffffe00052043c>] dump_stack+0x20/0x28
[    5.525038] [<ffffffe000031806>] panic+0xe0/0x242
[    5.529908] [<ffffffe000533900>] kernel_init+0xdc/0xe8
[    5.535168] [<ffffffe00002bf84>] ret_from_exception+0x0/0xc
[    5.540922] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel.
```
Full log: https://pastebin.com/p99rhbLk

jrrk, 2021-11-01:
Hi Pascal,
  I see nothing obviously wrong, but there could be an issue with the SD-Card you are are using being an untested model
or something went wrong with copying the files. To be honest I cannot now remember what card I used originally, I think
It was some “ultra-SD” brand.

Regards,
Jonathan

pcotret, 2021-11-01:
Ahh, I'm feeling dumb...
I used scripts given in the `lowrisc-quickstarter` repo.
**Solution:** partition for the root filesystem was too small for the `v0.7-rc2` release (https://github.com/lowRISC/lowrisc-quickstart/blob/master/Makefile#L63)

Can see my Ariane core now :+1: 
```bash
root@lowrisc:~# cat /proc/cpuinfo 
processor       : 0
hart            : 0
isa             : rv64imafdc
mmu             : sv39
uarch           : ethz,ariane
```

closed 2021-11-01 by pcotret