lowRISC/lowrisc-chip #205: Boot form SD and in emergency mode

helloworld-zzt, 2021-11-02:
when i boot from SD with debian  written in , it will enter emergency mode.   what i used is  genesys2_ariane_xilinx.new.mcs  and  rootfs.tar.xz  download from https://github.com/lowRISC/lowrisc-chip/releases

the message is as flowed:

[  132.544006] systemd[1]: Started Journal Service.
[  OK  ] Started Journal Service.
[  134.478956] EXT4-fs (rda2): re-mounted. Opts: (null)
[  OK  ] Started Remount Root and Kernel File Systems.
         Starting Flush Journal to Persistent Storage...
         Starting Load/Save Random Seed...
         Starting Create System Users...
[  143.211178] systemd-journald[90]: Received client request to flush runtime journal.
[  OK  ] Started Flush Journal to Persistent Storage.
[  OK  ] Started Create System Users.
         Starting Create Static Device Nodes in /dev...
[  OK  ] Started Create Static Device Nodes in /dev.
[  OK  ] Reached target Local File Systems (Pre).
         Starting udev Kernel Device Manager...
[  OK  ] Started udev Kernel Device Manager.
[ TIME ] Timed out waiting for device /dev/hvc0.
[DEPEND] Dependency failed for Serial Getty on hvc0.
[ TIME ] Timed out waiting for device /dev/rda1.
[DEPEND] Dependency failed for File System Check on /dev/rda1.
[DEPEND] Dependency failed for /mnt/dos.
[DEPEND] Dependency failed for Local File Systems.
[ TIME ] Timed out waiting for device /dev/rda4.
[DEPEND] Dependency failed for /home.
[DEPEND] Dependency failed for File System Check on /dev/rda4.
[ TIME ] Timed out waiting for device /dev/rda3.
[DEPEND] Dependency failed for /dev/rda3.
[DEPEND] Dependency failed for Swap.
[  OK  ] Stopped Dispatch Password …ts to Console Directory Watch.
[  OK  ] Stopped Forward Password R…uests to Wall Directory Watch.
[  OK  ] Reached target Timers.
         Starting Enable support fo…l executable binary formats...
[  OK  ] Reached target Login Prompts.
         Starting Preprocess NFS configuration...
[  OK  ] Closed Syslog Socket.
[  OK  ] Reached target Sockets.
[  OK  ] Started Emergency Shell.
[  OK  ] Reached target Emergency Mode.
         Starting Create Volatile Files and Directories...
[  OK  ] Started Enable support for…nal executable binary formats.
[  OK  ] Started Preprocess NFS configuration.
You are in emergency mode. After logging in, type "journalctl -xb" to view
system logs, "systemctl reboot" to reboot, "systemctl default" or "exit"
to boot into default mode.
Give root password for maintenance
(or press Control-D to continue): 
root@lowrisc:~# ls
root@lowrisc:~# cd [  395.732706] random: crng init done

jrrk, 2021-11-02:
Hi,
  systemd is a bit CPU hungry to run on this hardware. The problem you are having could be starting many parallel tasks and
not finishing them due to timeouts that are too short and/or lack of memory.

helloworld-zzt, 2021-11-03:
> Hi, systemd is a bit CPU hungry to run on this hardware. The problem you are having could be starting many parallel tasks and not finishing them due to timeouts that are too short and/or lack of memory.
> […](#)

Thanks for your suggestion!

helloworld-zzt, 2021-11-03:
> Hi, systemd is a bit CPU hungry to run on this hardware. The problem you are having could be starting many parallel tasks and not finishing them due to timeouts that are too short and/or lack of memory.
> […](#)
The FPGA board I used is Genesys2, whose memeory is  1GiB 1800Mt/s on-board DDR3. I guess the memory is enough. I want to know how to modify the timeout,and  in which file.  Looking forward to your reply.

jrrk, 2021-11-03:
> … <x-msg://4/#>

Hi,
  You can consult generic systems documentation for timeout control, e.g.:

https://newbedev.com/how-to-change-systemd-service-timeout-value <https://newbedev.com/how-to-change-systemd-service-timeout-value>

Unfortunately, it is a bit painful going through the debug loop to see the effect. You can mount the SD-card locally to make these modifications.

On the whole, I find it easier to use a light-weight system, which is why I went over to buildroot for this edition. Although the Genesis has plenty of
memory, it will still be crippled for CPU performance by the continuous task switching that systemd requires. Furthermore, the Linux options to run with
a slower tick rate are mostly disabled nowadays.

The Ariane core is about 15% slower than Rocket for unknown reasons.

helloworld-zzt, 2021-11-03:
> Hi, You can consult generic systems documentation for timeout control, e.g.: https://newbedev.com/how-to-change-systemd-service-timeout-value [...]

Thanks very much for your answer! The Rocket core can work easily. I will have a try as your suggestion.