lowRISC/lowrisc-chip #206: Source GNU Toolchain

Tiag0Carvalh0, 2021-12-15:
Hello 
So i am doing a project for my class and we need to use the RISC-V Open Source GNU Toolchain, to convert from C to RiscV, i am using **WSL**, version 20.04, and following this tutorial https://mindchasers.com/dev/rv-getting-started.
The problem is, when i do **"sudo make"** it executes, when i do **"sudo mkdir -p ~/Projects/riscv"**, no problems, but when i do "source **/opt/riscv32/env-riscv32"** it gives me an error **"-bash: /opt/riscv32/env-riscv32: No such file or directory"**, before these commands i have no problems. For what i can see my professor isnt very interested in helping, so thats why i am here.
Any thoughts?
Btw i am new at linux but i can handle it.
Thanks

jrrk2, 2021-12-16:
You should only ask on this forum issues to do with the LowRISC documentation and instructions, not about someone else’s tutorial. If someone else’s tutorial doesn’t work, you need to ask the author of that other tutorial. From your description, it sounds like the tutorial assumes RISCV tools are already installed. As a beginner, you should avoid using sudo in my opinion because it is rather easy to destroy your system. You can install a toolchain from here for example:

https://github.com/riscv-collab/riscv-gnu-toolchain

Tiag0Carvalh0, 2021-12-16:
Ok, thanks.
And sorry, new at this "environment"!

closed 2022-01-18 by jrrk