lowRISC/lowrisc-chip #207: Adding  a New uart can't boot the system

helloworld-zzt, 2021-12-26:
I am trying to add a new uart to the SOC (ariane-v0.7)with rocket shell. 
The top module add two logic signal :  input  logic  rx2  ,  output logic    tx2  , and the ariane_peripherals_xilinx module add the uart2 modle. 
uart_axi #(
    .AXI_ID_WIDTH   ( AxiIdWidth       ),
    .AXI_ADDR_WIDTH ( AxiAddrWidth     ),
    .AXI_DATA_WIDTH ( AxiDataWidth     ),
    .AXI_USER_WIDTH ( AxiUserWidth     ),
    .InclUART       ( InclUART         )
) i_uart_axi_UART2 (
    .clk_i  ( clk_i                    ),
    .rst_ni ( rst_ni                   ),
    .slave  ( master[ariane_soc::UART2] ),
    .rx_i   ( rx2_i                     ),
    .tx_o   ( tx2_o                     ),
    .cts_i  ( 1'b0                     ),
    .rts_o  (                          ),
    .irq_o  ( irq_sources[4]           )   //interrupt =<5> in device tree 
);  
In the ariane_soc_pkg.sv,  I have added the UART2 
typedef enum int unsigned {
    MOUSE    = 0,
    HID      = 1,
    GPIO     = 2,
    Ethernet = 3,
    SPI      = 4,
    UART     = 5,
    BOOT     = 6,
    UART2    = 7,   //UART2 hyh 12.16
    ExtLast  = 8
  } axi_extio_t;
and added "UART2Length    = 64'h10000;" , " UART2Base    = 64'h4700_0000 " .
IN the device tree, I also added 
uart: uart2@47000000 {
                        compatible = "ns16750";
                        reg = <0x46000000 0x1000>;
                        clock-frequency = <50000000>;
                        current-speed = <115200>;
                        interrupt-parent = <&L0>;
                        interrupts = <5>; 
                        reg-shift = <2>; // regs are spaced on 32 bit boundary
                        reg-io-width = <4>; // only 32-bit access are supported
                };

that's all i have changed, The  question is  

1. for  system verilog code of the soc,  what I have missed to change, something like parameter used in AXI ?
2. for booting the linux kernel , should I have to the add something since the original soc has a uart ?

Thanks for your attention.

helloworld-zzt, 2021-12-26:
Maybe the rocket core has only 4 interupts.

jrrk2, 2021-12-26:
If I remember rightly, this design already has a second UART for the Bluetooth keyboard/mouse interface. You could use this for your own purposes or copy the functionality. I can’t say why your version doesn’t work, most likely it is an invalid AXI bus transaction.

jrrk2, 2021-12-26:
If you need to expand the number of interrupts in Rocket this has to be done by modifying the Chisel generation scripts.

helloworld-zzt, 2022-01-02:
> 

Thanks for your reply. I did an invalid AXI bus transaction. I have corrected the mistake of the AXI BUS,  and changed the config of the numbers of interrupts, now it works.

closed 2022-01-18 by jrrk