lowRISC/lowrisc-chip #208: Recommended toolchain for  ethernet-v0.5

yswntht, 2022-09-06:
Hi,

What is the toolchain that you recommend for `git clone -b ethernet-v0.5 --recursive https://github.com/lowrisc/lowrisc-chip.git`.
The riscv-tools build that comes with clone has issue pulling binutils.2.26.tar.gz and thus build fails. The issue is consistent with all the releases that use binutils. Can you please point to riscv-tools that support ethernet-v0.5.

PS: I'm currently using ubuntu 18.04 with Vivid0 2015.4. Bitstream generation and XSim simulation with default boot.mem have not issue. I'm looking forward to try bare metal hello example. 

Thanks.

jrrk, 2022-09-06:
Thanks for your interest, this project has not been maintained for several years, nevertheless it should be possible to upgrade the version of riscv-tools that is called up by git, using the git submodule command. I don’t think I can give an accurate set of instructions by email but it should be entirely possible to make the changes yourself if you are sufficiently motivated. It is obviously out of my control if an external site stops hosting that particular version of binutils.

yswntht, 2022-09-06:
Thank you for quick response. I will work around the makefile with a different mirror of binuitls.