lowRISC/lowrisc-chip #209: Can't git clone

huahdgithub, 2023-08-02:
I've been trying to git clone the source code according to the documentation:
`git clone -b refresh-v0.6 --recursive https://github.com/lowrisc/lowrisc-chip.git lowrisc-chip-refresh-v0.6`
but keep running into these 2 error messages:
`fatal: remote error: upload-pack: not our ref 7ae23bc1974f8738b5009125873e43e6a8b1d848`
`fatal: Fetched in submodule path 'qemu', but it did not contain 7ae23bc1974f8738b5009125873e43e6a8b1d848. Direct fetching of that commit failed.`

Somebody help! THX!

jrrk, 2023-08-02:
This repository is not maintained for some while. It would be possible to get it going again, but you might prefer this one which is more up to date and supports symmetric multi-processing:

https://github.com/eugene-tarassov/vivado-risc-v.git

huahdgithub, 2023-09-13:
> This repository is not maintained for some while. It would be possible to get it going again, but you might prefer this one which is more up to date and supports symmetric multi-processing:
> 
> https://github.com/eugene-tarassov/vivado-risc-v.git

Thank you so much! Now I'm focusing on your suggested repo, with a new prblm as follows:
https://github.com/eugene-tarassov/vivado-risc-v/discussions/185
Would you offer some more advice? Thx again!

jrrk, 2023-09-13:
I see you are trying to build under WSL. Vivado is quite particular about the versions of Linux that it supports, check the supported O/S on Xilinx/AMD website. These O/S are fairly old, you might need to install a virtual machine or use cloud computing services. I tried this repo a few months ago and it worked fine.